micropython/micropython #7117: from rp2 import PIO, StateMachine, asm_pio -> Info -> Black Cat in a Coal Bunker

kevindawson, 2021-04-13:
- μpython v1.14 on 2021-04-09 (GNU 10.2.0 MinSizeRel)

from - Raspberry Pi Pico Python SDK (2021-04-07)
The idea is that for the 4 sets of pins ( in , out , set , sideset , excluding jmp ) that can be connected to a state machine,
there’s the following that need configuring for each set:

Sample code:
```python
from rp2 import PIO, StateMachine, asm_pio
from machine import Pin

@asm_pio(set_pins=(PIO.OUT_LOW, PIO.OUT_HIGH,  PIO.IN_LOW), sideset_pins=PIO.OUT_LOW)
def foo():
    pass

sm0 = StateMachine( 0, foo, freq=2000, set_pins=Pin(15), sideset_pins=Pin(22))
```
it just produces errors
```text
Traceback (most recent call last):
  File "<stdin>", line 5, in <module>
  File "rp2.py", line 230, in asm_pio
TypeError: unexpected keyword argument 'set_pins'
```
I can find info -> machine -> https://docs.micropython.org/en/latest/library/machine.html
but info for rp2 -> null

looking in the rp2 stubs - StateMachine
```python
def __init__(self, id, prog, freq: int=-1, *, in_base: Pin=None, out_base: Pin=None, set_base: Pin=None, jmp_pin: Pin=None, sideset_base: Pin=None, in_shiftdir: int=None, out_shiftdir: int=None, push_thresh: int=None, pull_thresh: int=None):
```
Ureka - moment - there NO set_pins in StateMachine instantiation
```python
from rp2 import PIO, StateMachine, asm_pio
from machine import Pin

@asm_pio(set_pins=(PIO.OUT_LOW, PIO.OUT_HIGH,  PIO.IN_LOW), sideset_pins=PIO.OUT_LOW)
def foo():
    pass

sm0 = StateMachine( 0, foo, freq=2000, set_base=Pin(15), out_base=Pin(16), sideset_base=Pin(22))
```
looking in the rp2 stubs - asm_pio
```python
def asm_pio(set_init: int = None, out_shiftdir: int = None, autopull: bool = None, pull_thresh: int = None, set_pins: Iterable[Sequence[int]] = None, sideset_pins: int = None, sideset_init: int = None, out_init: int = None, autopush: bool = None, push_thresh: int = None, in_base: int = None, out_base: int = None) -> Any:
```
Just hit the brick wall:(

Any chance someone that knows can put a working example up:
The idea is that for the 4 sets of pins ( in , out , set , sideset , excluding jmp ) that can be connected to a state machine,
there’s the following that need configuring for each set:

1. base GPIO
2. number of consecutive GPIO
3. initial GPIO direction (in or out pin)
4. initial GPIO value (high or low)

or do I need to add a documentation request -> Roadmap to next release v1.15

Thanks in Adv.

robert-hh, 2021-04-13:
It's set_base=Pin(x), sideset_base=Pin(x), in_base=Pin(x), out_base=Pin(x). But jmp_pin=Pin(x)
See also the examples in https://github.com/micropython/micropython/tree/master/examples/rp2.
The documentation is indeed missing.

kevindawson, 2021-04-13:
@robert-hh 
thanks for the conformation of the  StateMachine instantiation :)
I did look at the examples/rp2 - to no avail 
BUT 
the @asm_pio decorator has me vexed to say the least.

ps. it's years since I did Z80 - but I did have Zaks :)

kevindawson, 2021-04-13:
Another Ureka - moment

- ..._init in  @asm_pio decorator
- ..._base in StateMachine instantiation
```python
from rp2 import PIO, StateMachine, asm_pio
from machine import Pin

@asm_pio(set_init=PIO.OUT_LOW, out_init=PIO.OUT_HIGH, sideset_init=PIO.OUT_LOW)
def foo():
    pass

sm0 = StateMachine(
    0, foo, freq=2000, set_base=Pin(15), out_base=Pin(17), sideset_base=Pin(16)
)

sm0.active(1)
```
No Errors

@robert-hh
it's as if theirs an api change going on underneath

kevindawson, 2021-04-14:
@robert-hh 
#6933 I obversely did not look hard enough :)

lurch, 2021-04-14:
@kevindawson I _think_ the "logic" is that the `_init` parameters in the `@asm_pio` decorator specify both the number of pins, and also the initial value and direction of those pins at StateMachine startup. So `set_init=(PIO.OUT_LOW, PIO.OUT_HIGH,  PIO.IN_LOW)` says that the StateMachine has 3 "set" pins, initialised to (output, output, input) and (LOW, HIGH, LOW). And then when you construct a StateMachine and pass in `set_base=Pin(15)` it then "knows" to use pins 15, 16 and 17 as the 3 "set" pins.
And the `_init` parameters in the decorator are also "clever enough" to be able to be passed either a list (e.g. `set_init=(PIO.OUT_LOW, PIO.OUT_HIGH,  PIO.IN_LOW)`) to declare multiple pins, or a single value (e.g. `set_init=PIO.OUT_LOW`) to declare a single pin. But in all cases you only pass in a single pin to `set_base` in the `StateMachine` constructor.

kevindawson, 2021-04-14:
@lurch 
Thank you - I think that is the first Information I have read since I purchased a Pico for PI/O that makes sense with the correct parameters for both ```@asm_pio``` decorator and  ```StateMachine``` constructor.

But it is No Good, stuck in here (#7177) 

- The Pico μPython SDK needs to be kicked hard
- The μPython - rp2 - documentation needs to exist  #6832

lurch, 2021-04-14:
@kevindawson Glad I could help :smiley:   (my knowledge comes from my recollection of the private discussions that occurred during the development of the MicroPython rp2 port)

> The μPython - rp2 - documentation needs to exist

That's @dpgeorge 's responsibility.

But I _have_ just noticed that part of the problem is in the documentation created by Raspberry Pi, so I've just created a separate issue for that https://github.com/raspberrypi/pico-feedback/issues/142

aallan, 2021-04-15:
> > The μPython - rp2 - documentation needs to exist
>
> That's @dpgeorge 's responsibility.

I've added a note to the v1.15 release issue (#6832) linking to the issue (micropython#6855) talking about the rp2-port documentation. I agree it would be good for there to be more rp2-port documentation. However as @lurch says we view this as a responsibility of the MicroPython developer team rather than something Raspberry Pi is going to be doing going forward.

robert-hh, 2021-04-15:
So we have a few names and tasks:

- @dpgeorge Damien George for MicroPython
- @lurch Andrew Scheller for Raspberry Pi SDK(?), hardware(?)
- @aallan Alasdir Allan for the Raspberry Pi documentation
- @tjvr Tim Radvan, who told that he started to write the MicroPython PIO docs.

And obviously the whole community for testing, helping, misunderstanding and asking. Who else?

aallan, 2021-04-15:
> - @aallan Alasdir Allan for the Raspberry Pi documentation

While we're happy to accept PRs for the http://github.com/raspberrypi/pico-micropython-examples repo — and larger examples with the appropriate README.adoc files might well then get pulled into the [PDF documentation](https://datasheets.raspberrypi.org/pico/raspberry-pi-pico-python-sdk.pdf) in the App Notes appendix. We're not anticipating writing further general documentation of the MicroPython rp2-port beyond that.

There may be further [books](https://amzn.to/32hruYI) from Pi Press, however that's a separate proposition.

kevindawson, 2021-04-15:
@aallan @lurch 
Please correct me if I have got this wrong.

Raspberry Pi RP2040 New Pico - USP
  - μPython
  - pi/o

Documentation for **pi/o with μPython** may be forth coming from the Pi Press.

lurch, 2021-04-15:
[image]

:wink:

aallan, 2021-04-15:
> Please correct me if I have got this wrong.
> 
> Raspberry Pi RP2040 New Pico - USP

Sorry, what do you mean by "USP" here?

> μPython
> pi/o
> Documentation for pi/o with μPython may be forth coming from the Pi Press.

There is already PIO documentation in the current book, "[_Get Started with MicroPython on Raspberry Pi Pico_](https://hackspace.raspberrypi.org/books/micropython-pico)", which was published when Pico launched by Pi Press. There may well be further books coming out, but I have no insider knowledge of their publication schedule. I can't speak for them.

What I can say is that you should not anticipate further technical documentation beyond the existing "[_Raspberry Pi Pico Python SDK_](https://datasheets.raspberrypi.org/pico/raspberry-pi-pico-python-sdk.pdf)" book from Raspberry Pi (Trading) which was published at launch to accompany Pico. There is PIO documentation in that book also, which should be read alongside the [_"RP2040 Datasheet"_](https://datasheets.raspberrypi.org/rp2040/rp2040-datasheet.pdf) and [_"Raspberry Pi Pico C/C++ SDK"_](https://datasheets.raspberrypi.org/pico/raspberry-pi-pico-c-sdk.pdf) books which are the definitive sources of documentation (looking mainly at the RP2040 datasheet there) for PIO on RP2040.

GregWoods, 2021-08-19:
The C SDK Docs are 301 pages long. The MicroPython docs are 47 pages long and contains errors and inconsistencies, with nobody willing to do any work to improve the situation. There are PIO features which I can understand from the RP2040 docs, but knowing how MicroPython implements them  has involved going in the source files and reading C code, which defeats the point of using MicroPython.  As an example, try to find how to "pull" with "Blk" (block) set. Nothing in the MicroPython SDK docs. Reading https://github.com/micropython/micropython/blob/master/ports/rp2/rp2_pio.c  along with a little trial and error and I figured it out. But there is other stuff I can't. All in all I've found MicroPython ... particularly PIO on the rp2 a frustrating experience.

jimmo, 2021-08-20:
FYI there are in-progress docs for rp2 PIO at https://github.com/micropython/micropython/pull/7496  (This is part of MicroPython's participation in the Google Season of Docs project).

As you can imagine this is a complicated feature and not easy to document. Contributions welcome! Even if you don't feel like making a docs PR, just raising an issue with "I figured out this thing with PIO, might be useful to someone else" is worthwhile.