micropython/micropython #7118: ESP32 micropython Compilation error！

MRxing1, 2021-04-14:
CC extmod/mpexception.c
CC ../../lib/oofatfs/ff.c
CC ../../lib/oofatfs/ffunicode.c
CC ../../lib/littlefs/lfs2.c
CC ../../lib/littlefs/lfs2_util.c
CC ../../extmod/modbtree.c
In file included from ../../lib/berkeley-db-1.xx/PORT/include/../../btree/btree.h:44:0,
from ../../extmod/modbtree.c:38:
../../lib/berkeley-db-1.xx/PORT/include/mpool.h:50:2: error: expected specifier-qualifier-list before 'CIRCLEQ_ENTRY'
CIRCLEQ_ENTRY(_bkt) hq; /* hash queue /
^
../../lib/berkeley-db-1.xx/PORT/include/mpool.h:61:2: error: expected specifier-qualifier-list before 'CIRCLEQ_HEAD'
CIRCLEQ_HEAD(_lqh, _bkt) lqh; / lru queue head */
^
../../extmod/modbtree.c:329:14: error: unknown type name 'FILEVTABLE'
STATIC const FILEVTABLE btree_stream_fvtable = {
^
../../extmod/modbtree.c:330:5: error: initialization makes integer from pointer without a cast [-Werror=int-conversion]
mp_stream_posix_read,
^
../../extmod/modbtree.c:330:5: note: (near initialization for 'btree_stream_fvtable')
../../extmod/modbtree.c:331:5: error: excess elements in scalar initializer [-Werror]
mp_stream_posix_write,
^
../../extmod/modbtree.c:331:5: note: (near initialization for 'btree_stream_fvtable')
../../extmod/modbtree.c:332:5: error: excess elements in scalar initializer [-Werror]
mp_stream_posix_lseek,
^
../../extmod/modbtree.c:332:5: note: (near initialization for 'btree_stream_fvtable')
../../extmod/modbtree.c:333:5: error: excess elements in scalar initializer [-Werror]
mp_stream_posix_fsync
^
../../extmod/modbtree.c:333:5: note: (near initialization for 'btree_stream_fvtable')
../../extmod/modbtree.c: In function 'mod_btree_open':
../../extmod/modbtree.c:362:52: error: passing argument 2 of '__bt_open' makes integer from pointer without a cast [-Werror=int-conversion]
DB *db = __bt_open(MP_OBJ_TO_PTR(pos_args[0]), &btree_stream_fvtable, &open
^
In file included from ../../extmod/modbtree.c:37:0:
../../lib/berkeley-db-1.xx/PORT/include/db.h:230:5: note: expected 'int' but argument is of type 'const int *'
DB *__bt_open __P((const char *, int, int, const BTREEINFO *, int));
^
../../extmod/modbtree.c:362:75: error: passing argument 3 of '__bt_open' makes integer from pointer without a cast [-Werror=int-conversion]
*db = __bt_open(MP_OBJ_TO_PTR(pos_args[0]), &btree_stream_fvtable, &openinfo,
^
In file included from ../../extmod/modbtree.c:37:0:
../../lib/berkeley-db-1.xx/PORT/include/db.h:230:5: note: expected 'int' but argument is of type 'BTREEINFO * {aka struct *}'
DB *__bt_open __P((const char *, int, int, const BTREEINFO *, int));
^
../../extmod/modbtree.c:362:14: error: too few arguments to function '__bt_open'
DB *db = __bt_open(MP_OBJ_TO_PTR(pos_args[0]), &btree_stream_fvtable, &open
^
In file included from ../../extmod/modbtree.c:37:0:
../../lib/berkeley-db-1.xx/PORT/include/db.h:230:5: note: declared here
DB *__bt_open __P((const char *, int, int, const BTREEINFO *, int));
^
cc1: all warnings being treated as errors
../../py/mkrules.mk:63: recipe for target 'build-GENERIC_SPIRAM/extmod/modbtree.o' failed
make: *** [build-GENERIC_SPIRAM/extmod/modbtree.o] Error 1

micropytho Compilation error!
1 lib/berkeley-db-1.xx library file I have loaded and updated.
image
2 The following commands have been executed：
xkp@xkp-VirtualBox:/esp32/micropython$ git submodule init lib/berkeley-db-1.xx
xkp@xkp-VirtualBox:/esp32/micropython$ git submodule update
子模组路径 'lib/berkeley-db-1.xx'：检出 '7b31cae3c0892a7fca2089cb0354a4920c98c413'

The result is still not good!

Hope to get your reply!

Best wishes!

dpgeorge, 2021-04-14:
This could be related to symlinks in the `lib/berkeley-db-1.xx` directory.  What OS are you building under?  What is the contents of `lib/berkeley-db-1.xx/PORT/include/mpool.h`?

MRxing1, 2021-04-14:
1、my os is ubuntu-18.04.1
2 、The contents of lib/berkeley-db-1.xx/PORT/include/mpool.h is：
/*-
 * Copyright (c) 1991, 1993, 1994
 *	The Regents of the University of California.  All rights reserved.
 *
 * Redistribution and use in source and binary forms, with or without
 * modification, are permitted provided that the following conditions
 * are met:
 * 1. Redistributions of source code must retain the above copyright
 *    notice, this list of conditions and the following disclaimer.
 * 2. Redistributions in binary form must reproduce the above copyright
 *    notice, this list of conditions and the following disclaimer in the
 *    documentation and/or other materials provided with the distribution.
 * 3. All advertising materials mentioning features or use of this software
 *    must display the following acknowledgement:
 *	This product includes software developed by the University of
 *	California, Berkeley and its contributors.
 * 4. Neither the name of the University nor the names of its contributors
 *    may be used to endorse or promote products derived from this software
 *    without specific prior written permission.
 *
 * THIS SOFTWARE IS PROVIDED BY THE REGENTS AND CONTRIBUTORS ``AS IS'' AND
 * ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE
 * IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE
 * ARE DISCLAIMED.  IN NO EVENT SHALL THE REGENTS OR CONTRIBUTORS BE LIABLE
 * FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL
 * DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS
 * OR SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION)
 * HOWEVER CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT
 * LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY
 * OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF
 * SUCH DAMAGE.
 *
 *	@(#)mpool.h	8.2 (Berkeley) 7/14/94
 */

#include <sys/queue.h>

/*
 * The memory pool scheme is a simple one.  Each in-memory page is referenced
 * by a bucket which is threaded in up to two of three ways.  All active pages
 * are threaded on a hash chain (hashed by page number) and an lru chain.
 * Inactive pages are threaded on a free chain.  Each reference to a memory
 * pool is handed an opaque MPOOL cookie which stores all of this information.
 */
#define	HASHSIZE	128
#define	HASHKEY(pgno)	((pgno - 1) % HASHSIZE)

/* The BKT structures are the elements of the queues. */
typedef struct _bkt {
	CIRCLEQ_ENTRY(_bkt) hq;		/* hash queue */
	CIRCLEQ_ENTRY(_bkt) q;		/* lru queue */
	void    *page;			/* page */
	pgno_t   pgno;			/* page number */

#define	MPOOL_DIRTY	0x01		/* page needs to be written */
#define	MPOOL_PINNED	0x02		/* page is pinned into memory */
	u_int8_t flags;			/* flags */
} BKT;

typedef struct MPOOL {
	CIRCLEQ_HEAD(_lqh, _bkt) lqh;	/* lru queue head */
					/* hash queue array */
	CIRCLEQ_HEAD(_hqh, _bkt) hqh[HASHSIZE];
	pgno_t	curcache;		/* current number of cached pages */
	pgno_t	maxcache;		/* max number of cached pages */
	pgno_t	npages;			/* number of pages in the file */
	u_long	pagesize;		/* file page size */
	int	fd;			/* file descriptor */
					/* page in conversion routine */
	void    (*pgin) __P((void *, pgno_t, void *));
					/* page out conversion routine */
	void    (*pgout) __P((void *, pgno_t, void *));
	void	*pgcookie;		/* cookie for page in/out routines */
#ifdef STATISTICS
	u_long	cachehit;
	u_long	cachemiss;
	u_long	pagealloc;
	u_long	pageflush;
	u_long	pageget;
	u_long	pagenew;
	u_long	pageput;
	u_long	pageread;
	u_long	pagewrite;
#endif
} MPOOL;

__BEGIN_DECLS
MPOOL	*mpool_open __P((void *, int, pgno_t, pgno_t));
void	 mpool_filter __P((MPOOL *, void (*)(void *, pgno_t, void *),
	    void (*)(void *, pgno_t, void *), void *));
void	*mpool_new __P((MPOOL *, pgno_t *));
void	*mpool_get __P((MPOOL *, pgno_t, u_int));
int	 mpool_put __P((MPOOL *, void *, u_int));
int	 mpool_sync __P((MPOOL *));
int	 mpool_close __P((MPOOL *));
#ifdef STATISTICS
void	 mpool_stat __P((MPOOL *));
#endif
__END_DECLS

dpgeorge, 2021-04-14:
Ok, so it's not an issue with symlinks.

What IDF version are you using?  Did you install the correct xtensa toolchain?

MRxing1, 2021-04-14:
Building with ESP IDF v4
https://github.com/espressif/esp-idf
branch:release/v4.0

[image]
[image]

I think my toolchain is installed properly。

MRxing1, 2021-04-14:
The log：
xkp@xkp-VirtualBox:~/esp32/micropython/ports/esp32$ make clean
MicroPython v1.13-30-g1d344f446-dirty
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Building with ESP IDF v4
rm -rf build-GENERIC_SPIRAM 
xkp@xkp-VirtualBox:~/esp32/micropython/ports/esp32$ make 
MicroPython v1.13-30-g1d344f446-dirty
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Building with ESP IDF v4
GEN build-GENERIC_SPIRAM/sdkconfig.h
mkdir -p build-GENERIC_SPIRAM/bootloader//home/xkp/esp32/esp-idf/components/bootloader/subproject/main/
mkdir -p build-GENERIC_SPIRAM/bootloader//home/xkp/esp32/esp-idf/components/bootloader_support/src/
mkdir -p build-GENERIC_SPIRAM/bootloader//home/xkp/esp32/esp-idf/components/bootloader_support/src/esp32/
mkdir -p build-GENERIC_SPIRAM/bootloader//home/xkp/esp32/esp-idf/components/log/
mkdir -p build-GENERIC_SPIRAM/bootloader//home/xkp/esp32/esp-idf/components/soc/esp32/
mkdir -p build-GENERIC_SPIRAM/bootloader//home/xkp/esp32/esp-idf/components/soc/src/
mkdir -p build-GENERIC_SPIRAM/bootloader//home/xkp/esp32/esp-idf/components/spi_flash/
CC /home/xkp/esp32/esp-idf/components/bootloader_support/src/bootloader_clock.c
CC /home/xkp/esp32/esp-idf/components/bootloader_support/src/bootloader_common.c
CC /home/xkp/esp32/esp-idf/components/bootloader_support/src/bootloader_flash.c
CC /home/xkp/esp32/esp-idf/components/bootloader_support/src/bootloader_flash_config.c
CC /home/xkp/esp32/esp-idf/components/bootloader_support/src/bootloader_init.c
CC /home/xkp/esp32/esp-idf/components/bootloader_support/src/bootloader_random.c
CC /home/xkp/esp32/esp-idf/components/bootloader_support/src/bootloader_utility.c
CC /home/xkp/esp32/esp-idf/components/bootloader_support/src/flash_qio_mode.c
CC /home/xkp/esp32/esp-idf/components/bootloader_support/src/esp_image_format.c
CC /home/xkp/esp32/esp-idf/components/bootloader_support/src/flash_encrypt.c
CC /home/xkp/esp32/esp-idf/components/bootloader_support/src/flash_partitions.c
CC /home/xkp/esp32/esp-idf/components/bootloader_support/src/esp32/bootloader_sha.c
CC /home/xkp/esp32/esp-idf/components/bootloader_support/src/esp32/secure_boot.c
AR build-GENERIC_SPIRAM/bootloader/libbootloader_support.a
CC /home/xkp/esp32/esp-idf/components/log/log.c
AR build-GENERIC_SPIRAM/bootloader/liblog.a
CC /home/xkp/esp32/esp-idf/components/spi_flash/spi_flash_rom_patch.c
AR build-GENERIC_SPIRAM/bootloader/libspi_flash.a
CC /home/xkp/esp32/esp-idf/components/soc/esp32/cpu_util.c
CC /home/xkp/esp32/esp-idf/components/soc/esp32/gpio_periph.c
CC /home/xkp/esp32/esp-idf/components/soc/esp32/rtc_clk.c
CC /home/xkp/esp32/esp-idf/components/soc/esp32/rtc_clk_init.c
CC /home/xkp/esp32/esp-idf/components/soc/esp32/rtc_init.c
CC /home/xkp/esp32/esp-idf/components/soc/esp32/rtc_periph.c
CC /home/xkp/esp32/esp-idf/components/soc/esp32/rtc_pm.c
CC /home/xkp/esp32/esp-idf/components/soc/esp32/rtc_sleep.c
CC /home/xkp/esp32/esp-idf/components/soc/esp32/rtc_time.c
CC /home/xkp/esp32/esp-idf/components/soc/esp32/rtc_wdt.c
CC /home/xkp/esp32/esp-idf/components/soc/esp32/sdio_slave_periph.c
CC /home/xkp/esp32/esp-idf/components/soc/esp32/sdmmc_periph.c
CC /home/xkp/esp32/esp-idf/components/soc/esp32/soc_memory_layout.c
CC /home/xkp/esp32/esp-idf/components/soc/esp32/spi_periph.c
CC /home/xkp/esp32/esp-idf/components/soc/src/memory_layout_utils.c
AR build-GENERIC_SPIRAM/bootloader/libsoc.a
CC /home/xkp/esp32/esp-idf/components/bootloader/subproject/main/bootloader_start.c
AR build-GENERIC_SPIRAM/bootloader/libmain.a
LINK build-GENERIC_SPIRAM/bootloader.elf
Create build-GENERIC_SPIRAM/bootloader.bin
esptool.py v2.9-dev
Create build-GENERIC_SPIRAM/partitions.bin
mkdir -p build-GENERIC_SPIRAM/genhdr
GEN build-GENERIC_SPIRAM/genhdr/mpversion.h
GEN build-GENERIC_SPIRAM/genhdr/moduledefs.h
GEN build-GENERIC_SPIRAM/genhdr/qstr.i.last
GEN build-GENERIC_SPIRAM/genhdr/qstr.split
GEN build-GENERIC_SPIRAM/genhdr/qstrdefs.collected.h
QSTR updated
GEN build-GENERIC_SPIRAM/genhdr/qstrdefs.generated.h
GEN build-GENERIC_SPIRAM/genhdr/compressed.split
GEN build-GENERIC_SPIRAM/genhdr/compressed.collected
Compressed data updated
GEN build-GENERIC_SPIRAM/genhdr/compressed.data.h
[... 263 lines cut ...]
../../extmod/modbtree.c: In function 'mod_btree_open':
../../extmod/modbtree.c:362:52: error: passing argument 2 of '__bt_open' makes integer from pointer without a cast [-Werror=int-conversion]
     DB *db = __bt_open(MP_OBJ_TO_PTR(pos_args[0]), &btree_stream_fvtable, &openinfo, /*dflags*/ 0);
                                                    ^
In file included from ../../extmod/modbtree.c:37:0:
../../lib/berkeley-db-1.xx/PORT/include/db.h:230:5: note: expected 'int' but argument is of type 'const int *'
 DB *__bt_open __P((const char *, int, int, const BTREEINFO *, int));
     ^
../../extmod/modbtree.c:362:75: error: passing argument 3 of '__bt_open' makes integer from pointer without a cast [-Werror=int-conversion]
     DB *db = __bt_open(MP_OBJ_TO_PTR(pos_args[0]), &btree_stream_fvtable, &openinfo, /*dflags*/ 0);
                                                                           ^
In file included from ../../extmod/modbtree.c:37:0:
../../lib/berkeley-db-1.xx/PORT/include/db.h:230:5: note: expected 'int' but argument is of type 'BTREEINFO * {aka struct <anonymous> *}'
 DB *__bt_open __P((const char *, int, int, const BTREEINFO *, int));
     ^
../../extmod/modbtree.c:362:14: error: too few arguments to function '__bt_open'
     DB *db = __bt_open(MP_OBJ_TO_PTR(pos_args[0]), &btree_stream_fvtable, &openinfo, /*dflags*/ 0);
              ^
In file included from ../../extmod/modbtree.c:37:0:
../../lib/berkeley-db-1.xx/PORT/include/db.h:230:5: note: declared here
 DB *__bt_open __P((const char *, int, int, const BTREEINFO *, int));
     ^
cc1: all warnings being treated as errors
../../py/mkrules.mk:63: recipe for target 'build-GENERIC_SPIRAM/extmod/modbtree.o' failed
make: *** [build-GENERIC_SPIRAM/extmod/modbtree.o] Error 1

dpgeorge, 2021-04-14:
You'll need IDF version v4.0.2 or later.  Try `v4.2`.

MRxing1, 2021-04-14:
OK ,i will try,  best wishes !

MRxing1, 2021-04-14:
I tried 4.2 and it still didn't work

xkp@xkp-VirtualBox:~/esp32/micropython/ports/esp32$ make clean
MicroPython v1.13-30-g1d344f446-dirty
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Building with ESP IDF v4
rm -rf build-GENERIC_SPIRAM 
xkp@xkp-VirtualBox:~/esp32/micropython/ports/esp32$ make
MicroPython v1.13-30-g1d344f446-dirty
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Building with ESP IDF v4
GEN build-GENERIC_SPIRAM/sdkconfig.h
Traceback (most recent call last):
  File "/home/xkp/esp32/esp-idf/tools/kconfig_new/confgen.py", line 641, in <module>
    main()
  File "/home/xkp/esp32/esp-idf/tools/kconfig_new/confgen.py", line 261, in main
    config = kconfiglib.Kconfig(args.kconfig)
  File "/home/xkp/esp32/esp-idf/tools/kconfig_new/kconfiglib.py", line 947, in __init__
    self._init(filename, warn, warn_to_stderr, encoding)
  File "/home/xkp/esp32/esp-idf/tools/kconfig_new/kconfiglib.py", line 1085, in _init
    self._parse_block(None, self.top_node, self.top_node).next = None
  File "/home/xkp/esp32/esp-idf/tools/kconfig_new/kconfiglib.py", line 2978, in _parse_block
    if self.srctree else "unset or blank"))
kconfiglib.KconfigError: /home/xkp/esp32/esp-idf/Kconfig:165: '$COMPONENT_KCONFIGS_PROJBUILD_SOURCE_FILE' not found (in 'source "$COMPONENT_KCONFIGS_PROJBUILD_SOURCE_FILE"'). Check that environment variables are set correctly (e.g. $srctree, which is unset or blank). Also note that unset environment variables expand to the empty string.
Makefile:411: recipe for target 'build-GENERIC_SPIRAM/sdkconfig.h' failed
make: *** [build-GENERIC_SPIRAM/sdkconfig.h] Error 1
xkp@xkp-VirtualBox:~/esp32/micropython/ports/esp32$

dpgeorge, 2021-04-14:
> MicroPython v1.13-30-g1d344f446-dirty

This is an old version.  You need to use the latest version of MicroPython.

MRxing1, 2021-04-14:
OK, I see. Thank you very much. 
It is suggested to add support for queue in the new version of micropython. My version does not have the function of queue, so it is all added by myself.

Thank you very much!
Best wishes!

YanMinge, 2021-04-21:
I also encountered some compilation problems, and would like to know what version of cmake you are using?

https://forum.micropython.org/viewtopic.php?f=2&t=10103

dpgeorge, 2021-04-22:
> I also encountered some compilation problems?

Those issues on the forum seem related to the CMake version, which needs to be 3.12 at a minimum to get Python3.  See also #7003.

closed 2021-04-23 by dpgeorge