micropython/micropython #7120: CAN.RECV  micropython

LASERXPRESS, 2021-04-14:
Hi 
We have a problem to receive any data on CAN1 or CAN2  .

We can transmit data out  - working 100 %  

If we send data  to the can port  -  we can see on the tx line the acknowledge pule is sent out  by the pyboard .  That means the data is received by the can controller  - else the sender unit will keep on transmitting .

The problem is that we can not access the data  ?  

We tried all the sample files   on microphthon.org but no success     . 

what are we missing   ?  Can somebody please help with a  small tested working sample to receive data on 250kbs .
For now we do not need any filters  - any data should be received .

This is not the first time we use can bus  , but the first time on the pyboard .   We have all the required bus drivers etc  in place .

We use micropython  v1.14 -153-g22554cf8e       date  12 apr 2021    on a pyboard   v1,1

kind regards
nico

dpgeorge, 2021-04-14:
Can you please post the (minimal) Python code that you run which is doing the `recv`?

LASERXPRESS, 2021-04-14:
Thank you for the quick response!

Here's the basic code I tried to get the Can working. I am not sure how to print code in the correct format on the post but there is a text file with the code in as well, the program compiles and run smoothly on the pyboard, no errors, just not receiving anything on can bus

[pyboardCanExample.txt](https://github.com/micropython/micropython/files/6309735/pyboardCanExample.txt)

import _thread
from pyb import CAN

can = CAN(1, CAN.NORMAL, baudrate=250000)
pyb.delay(1000);
can.send('message!', 123)
pyb.delay(1000);
#can.recv(0)

def canRead():
    while can.any(0):
        data = can.recv(0, timeout=10000)
        print(data)
        print("canRead")

t3 = _thread.start_new_thread(canRead,())

Thank you

dpgeorge, 2021-04-14:
Can you try:
- without using `_thread`
- by enabling a filter: `can.setfilter(0, CAN.MASK16, 0, (0, 0, 0, 0))`
- using `CAN.LOOPBACK` mode instead of `CAN.NORMAL`

LASERXPRESS, 2021-04-15:
HI George
thanks for the reply.
  
We need to use the _thread option -  nature of the program - but will test without it
Will test with that filter setting  
if we use loopback we just get what was sent out  - so can you please clearly detail the difference between the   ,normal ,loopback etc   options  .

Kind regards
Nico

LASERXPRESS, 2021-04-15:
Hi

We tested the 3 things you asked us to try but still nothing is read on the can.

I have posted a screenshot of the terraterm screen where it printed the message that was received on the canbus but that is the message that I sent out from the pyboard and it just reads it back in LOOPBACK mode, in NORMAL mode it does not read the message back that it just sent. I can not send a message from another can device and receive it with the pyboard.
[image]

I have posted another screenshot showing canHacker where we read the message in that the pyboard sends on the top and on the bottom the message that I try to send back to the pyboard but its not receiving the message.

[image]

[pyboardCanExample.txt](https://github.com/micropython/micropython/files/6315643/pyboardCanExample.txt)

Just before I posted this we decided to load the firmware without threading as well because we tested the above without _thread library but we still had the threading firmware. Now the can messages can be received in the pyboard that was sent from the canHacker.

[image]

We unfortunately need to use threading in the final program. If you maybe have a solution for using the threading and still be able to send and receive can messages that would be great but for now we can carry on.

Thank you for the help

LASERXPRESS, 2021-04-15:
Hi 
On a different  topic  , but sort of related   ,  can the      xtsd04glgeag     (    https://blog.adafruit.com/2021/01/27/new-product-adafruit-spi-flash-sd-card-xtsd-512-mb/   )   be used instead of an external  sd card   ?   The on chip memory might not be enough so I want to add external flash but prefer it  fixed to the board  .  

I might just add a link for the card detect switch to select   in-cpu storage or  external flash chip storage  . 

Kind regards

LASERXPRESS, 2021-04-15:
Hi

Problem Solved !

There was a error in the program. I started the def canRead with the threading and inside of the def there is a while loop, but the while loop at the moment that the threading called it was false because there is nothing on the canbus to read so the while runs out and the def ends. I have now just added another while True: loop around the while can.any(0): loop and it stays inside the while loop and I can read messages on the canbus.

Here is the updated code, this is now with the newest threading firmware and using _thread as well to call the def.

[pyboardCanExampleThread.txt](https://github.com/micropython/micropython/files/6317300/pyboardCanExampleThread.txt)

break11, 2022-01-14:
Hi all, i use pyboard v1.1 with MCV2551 transciever. I do not understand how works mode pyb.CAN.LOOPBACK. In my case in this mode i receive on pyboard only that messages, which i send from pyboard. But i thinked what in CAN.LOOPBACK i can receive all messages inlude that messages sended from pyboard and all other. In  pyb.CAN.NORMAL mode i receive all messages exclude message which was sending from pyboard.

jonnor, 2024-09-13:
The original issue reported by @LASERXPRESS was resolved. Closing.

@break11 please open a separate discussion thread in our QA section if your question is still relevant. https://github.com/orgs/micropython/discussions/categories/core-development

closed 2024-09-13 by jonnor