micropython/micropython #7124: rp2: crash and memory corruption with threading

robert-hh, 2021-04-16:
In a use case, where two threads are created and massive UART write happens, various kinds of crashes and memory corruptions occur. Test script:
```
from machine import UART, Pin
import time
import _thread

uart = UART(1, 115200 , parity=None, stop=1, bits=8, tx=Pin(4), rx=Pin(5), txbuf=32, timeout=10)

def send(command):
        write_buffer = bytearray(len(command) + 3)
        write_buffer[0: len(command)] = bytes(command, 'ASCII')
        write_buffer[-3:] = b'\xff\xff\xff'
        uart.write(write_buffer)

def burst():
    msg="0123456789" * 8 + "\r\n"
    n=0
    while True:
        uart.write("%d: " % n)
        uart.write(msg)
        n += 1

def count_count():
    n = 1
    while True:
        print("Count Count counts", n)
        n += 1
        time.sleep(1)

_thread.start_new_thread(burst, ())

# burst()
count_count()
```
The function send() is not called, but the behavior changes on whether is is present in the code or commented out. For the test, UART1 tx is connected with a UART/USB bridge to a PC showing the data stream. The following test set-ups were made:

- With send() code, thread 0 REPL, thread 1 burst(): 
         Result: memory allocation error at msg 5721, rare character doubling, like:
                "MemoryError: memory allocation failed, allocating 875706184 bytes"
                "5475: 011223344556677889900112233456789012345678901234567890123456789012345678901234567890123456789"
- With send() code, thread 0 count_count(), thread 1 burst(): 
        Result: Lock-up after msg 5721, rare char doubling
- Without send() code, thread 0 count_count(), thread 1 burst():
        Result: No lock-up, rare char doubling
- Without send() code, thread 0 REPL, , thread 1 burst()
        Result: No lock-up, no transmission errors

- With send() code, thread 0 burst(), thread 1 count_count()
        Result: sending complete rubbish after msg 5720, like:
                "43: h5��� 44: h5�� 45: h5�0� 46: h5�P� 47: h5�p� 48: h5��� 49: h5��� 50: h5��� 51: h5{�� 52: h5��" and so on
- Without send() code, thread 0 burst(), thread 1 count_count()
        Result No problem. stopped after msg 100_000

Using critical section instead of lock does not change the game (not surprising).
Increasing the thread stack size has no effect (expected).

dpgeorge, 2021-04-16:
My first guess would be that it's something to do with the GC interacting between the two cores/threads.  Can you try writing a test that does not do any memory allocations when running (ie only at start up)?

robert-hh, 2021-04-16:
Good guess, indeed. Without allocation it looks much better. Same test, 100_000 cycles w/o crash and strange output only in one mode.
I modified the test code such that the increasing numbers are hand-coded into a bytearray, and sys.stdout.write() is used instead of print.
- thread0 REPL, thread1 burst(): No crash, no strange output. stopped at 100_00 cycles
- thread0 count_count(), thread1 burst(): No crash, char doubling about every 1000th message, stopped at 50_000 cycles
- thread0 burst, thread 1 count_count():  No crash, no strange output. stopped at 50_00 cycles

robert-hh, 2021-04-17:
The character doubling in UART only occurs, if data is sent to stdout at the same time in thread 0. It does not matter if doing so by calling print() or by sys.stdout.write(). If I just let the LED blink, or interleave writing to stdout and to UART, everything works well.
interestingly, this character doubling only happens, if UART write is done in thread 1 and stdout write in thread 0. If the threads are swapped, it's fine.

robert-hh, 2021-04-17:
I tried to figure out, whether the twice sent characters weer actually put into the UART FIFO by the code in uart_fill_tx_fifo(). So I modified the message to not set the same character twice in sequence, and checked in the uart_fill_tx_fifo() function, whether the actual character is the same as the previous one (code below). If yes, it would create a short pulse at a GPIO pin, which I then would be able to detect. This trigger happened much more rare that the character doubling. And that puzzles me. In a sequence of 100_000 messages, I had 74 with character duplications, but only one trigger event. b.t.w.: If the trigger happens, the function uart_fill_tx_fifo() was called in a interrupt context.

Modified uart_fill_tx_fifo(). P11 was set in the Python code as output. If the pulse is generated when two consecutive chars are different, it fires well.
```
uint32_t *PIO_OUT = (uint32_t *)0xd0000010;

// take bytes from the buffer and put them into the UART FIFO
STATIC void uart_fill_tx_fifo(machine_uart_obj_t *self) {
    static int last_char = 0;
    while (uart_is_writable(self->uart) && ringbuf_avail(&self->write_buffer) > 0) {
        // get a byte from the buffer and put it into the uart
        int c = ringbuf_get(&(self->write_buffer));
        if (c == last_char) {
            PIO_OUT[0] |= 1 << 11; // set P11
        }
        last_char = c;
        uart_get_hw(self->uart)->dr = c;
        PIO_OUT[0] &= ~(1 << 11);  // clear p11
    }
}
```

peterhinch, 2021-04-18:
> That all looks like the order of execution is disturbed.

A high priority interrupt? Can recursive execution of the same ISR occur?

robert-hh, 2021-04-18:
I thought about stacked ISR calls. But that should affect the sequence of bytes in the output. But he order is unchanged, just some bytes are sent twice in a row.

robert-hh, 2021-04-18:
After increasing the output rate in thread0, the pulse width of the trigger pulse created in thread 1 is back to ~60ns. Very strange. And sometimes the ISR seems to last for more than 15ms, falling back to its usual ~20µs soon later. So it seems there is a lot of trouble with ISR and threading.

robert-hh, 2021-04-28:
**Partially solved.** After having put aside this problem for a while, I had some progress now. If UART instatiation and UART use happen in the same thread, then the character doubling in the output does not happen.

The crash/lock-up which occurs when memory allocations are present is still present.

dpgeorge, 2021-05-04:
The fix #7217 may be related to this, but not quite sure yet.  If you have a chance to test that PR with the issue here that would be helpful.

robert-hh, 2021-05-04:
Sorry, it does not fix either of the two problems, I see here. It still stops after a while, if memory is allocated in the second thread. And if not, it still send character doubles if uart IRQ handler and uart send are not located in the same thread.

jrullan, 2021-12-01:
I too have experienced similar issues in my project.  I documented my findings in the raspberry pi pico forum. 
https://forums.raspberrypi.com/viewtopic.php?t=324136

peterhinch, 2021-12-01:
I wonder if https://github.com/micropython/micropython/issues/7977 is relevant. Interestingly that problem isn't restricted to UART output: SPI behaves identically.

jrullan, 2021-12-01:
I'll repost here what I posted in the Raspberry Pi Pico forum in the hopes that this issue might be taken over again. The project I am attaching **does work**, but with several hacks and workarounds, none of which are fully characterized. The most strange/bizarre issue is the one requiring the "padding" array, look in the code below for MEMORY_ALIGNMENT_ARRAY.

> The culprit seems to be the implementation of the _thread module for the RP2. I noticed several things.
> 
> First it seems that the memory allocation done for the second thread somehow causes the garbage collector to stop working > unless specifically told to by using gc.collect() at specific places in the code, where memory leaks if and when the second thread is started.
> 
> Second, it seems that there is an issue of overlapping memory areas when the second thread is started because in order to make my code work I had to declare a block of memory in the form of an array of integers just before starting the second thread or otherwise the code stalls. Declaring the same array after starting the thread does not work and also the amount of integers is significant. I managed to make it work with any quantity above 512 integers which I think is related to the two 256 arrays I use to manipulate the LEDs.
> 
> Third, all the variables needed to be declared globally to prevent memory leakages.
> 
> I found a couple of posts on the Micropython forums that describe similar issues, so it seems the rp2 implementation of Micropython still needs work to iron out the memory issues.
> 
> Some links that have similar issues:
> 
> https://forum.micropython.org/viewtopic ... 21&t=10331
> https://forum.micropython.org/viewtopic ... lit=thread
> https://forum.micropython.org/viewtopic ... ead#p62348
> https://github.com/micropython/micropython/issues/7124

This is the code I used in my project, maybe it is useful to replicate and debug the problem. 

``` python
import array, utime, _thread, gc
from machine import Pin, UART
import rp2

try: 
    import gfx
except ImportError:
    raise RuntimeError("Cannot find gfx. Have you copied gfx.py to your Pico?")

#================
#   Objects
#================

# Serial object for the HC-05 communications
uart = UART(0,9600)

# Graphics object used to draw primitives
def set_pixel_xy(x,y,color,brightness=1):
    # Set a pixel using its x,y position (inverting row, column)
    set_pixel_rc(y,x,color,brightness)
    
graphics = gfx.GFX(32,8,set_pixel_xy)

#================
#   Constants
#================

# Matrix Configuration
STRIPS = 1
SUBMATRIX_HEIGHT = 8
SUBMATRIX_WIDTH = 32

# Configure the number of WS2812 LEDs.
NUM_LEDS = 256
PIN_NUM = 15

MAX_CYCLES = 5

#================
# Global Variables
#================
brightness = 0.1
color0 = 0
color1 = 0
color2 = 0
r=0
g=0
b=0
[... 506 lines cut ...]
        sb = 0
        if y1 == y2:
            last = y1
        else:
            last = y1-1
        for y in range(y0, last+1):
            a = x0 + sa // dy01
            b = x0 + sb // dy02
            sa += dx01
            sb += dx02
            if a > b:
                a, b = b, a
            self.hline(a, y, b-a+1, *args, **kwargs)
        sa = dx12 * (y - y1)
        sb = dx02 * (y - y0)
        while y <= y2:
            a = x1 + sa // dy12
            b = x0 + sb // dy02
            sa += dx12
            sb += dx02
            if a > b:
                a, b = b, a
            self.hline(a, y, b-a+1, *args, **kwargs)
            y += 1
```

helarsen, 2022-02-13:
Allow me to piggy back onto this thread with similar findings of corruption when using _thread and UART on RPi Pico. Reported here.
[https://forums.raspberrypi.com/viewtopic.php?t=329639](url)
I can provide the complete code if it makes sense.

yjchun, 2022-02-16:
I was testing this issue and found something interesting.
With following two modifications, I could run above test code successfully (at least for 30 minutes).

1. modify ports/rp2/mpthreadport.c:59
```python
            //gc_collect_root((void **)core1_stack, core1_stack_num_words);
            gc_collect_root((void **)&core1_stack, core1_stack_num_words);
```

2. main.py: put gc.collect() in main loop.
```python
from machine import UART, Pin
import time
import _thread
import gc

uart = UART(0, 115200 , parity=None, stop=1, bits=8, tx=Pin(0), rx=Pin(1), txbuf=32, timeout=10)

def send(command):
        write_buffer = bytearray(len(command) + 3)
        write_buffer[0: len(command)] = bytes(command, 'ASCII')
        write_buffer[-3:] = b'\xff\xff\xff'
        uart.write(write_buffer)

def burst():
    msg="0123456789" * 8 + "\r\n"
    n=0
    while True:
        uart.write("%d: " % n)
        uart.write(msg)
        n += 1

def count_count():
    n = 1
    while True:
        print(gc.mem_free())
        gc.collect()
        print("Count Count counts", n)
        n += 1
        time.sleep(1)

_thread.start_new_thread(burst, ())
count_count()
```
With these two modifications, it ran successfully.
My test results: 
1. I couldn't explain why this works reading the code: `gc_collect_root((void **)&core1_stack, core1_stack_num_words);`
2. without gc.collect() in main thread, the moment GC is out of memory is when thread stops.
3. gc.collect() in thread stop the thread, same reason with (2)
4. I did notice character doubling problem very rarely in test cases. I am using master rev. as of today. Instead, without modification (1), uart is sending some garbages when burst() is called from main thread: `930: €Rå0 931: €RdP 932: €Rçp`. Only heap data is lost.

After tests, my guess is that GC information (or address) is not properly setup in `core1`. I will try to look into it later but I am not familiar with micropython source code yet.

yjchun, 2022-02-16:
A quick note on character doubling problem: When I was reading rp2/machine_uart.c, It was suspicious that write_buffer and read_buffer is not properly protected against IRQ: write_lock and read_lock to be set for all access to read/write_buffer? This might be the cause but didn't verify yet though.

dpgeorge, 2022-03-01:
Fixed by 2d47020e15b82b7f021e0b52dabe75baf5cb4c4d

closed 2022-03-01 by dpgeorge