micropython/micropython #7126: utime: Syntax error <function> not reported.

chrismas9, 2021-04-17:
if `ticks_ms` instead of `ticks_ms()` is used inside `ticks_add` or `ticks_diff`
`<function>` syntax error is not reported.

[image]

dpgeorge, 2021-04-18:
This is essentially the same as #7115.

It's definitely not a syntax error because there is nothing illegal about the syntax.  It could be a type error because ticks add/diff only accepts small integers and above it is passed a function.  But (see #7115) the idea with these ticks functions is that they are efficient so no checks are done (and it's still safe, cannot crash).

closed 2021-04-18 by chrismas9

tongdee9070, 2021-04-18:
The info is goo but too over welming, thank you