micropython/micropython #7127: stm32/boards: Add STM32H750VBT6 board.

THSLP13, 2021-04-17:
Now I have a board based on STM32H750 MCUs.
It has a 64 Mbit external flash.
So now I confused how I can build a firmware for it,it only have 128 kb ROM.
I can't write a bootloader to jump to app because I have poor programming skills.(And I have experienced several compile failure).
So why not provide a new port for STM32H750VBT6?It's so fast.
I did my best,then I post an issue here to ask for help.
(And there is something wrong with my msys2...why my Python cannot  work???)

(Poor English ORZ...)

THSLP13, 2021-04-17:
And there is an SD port that I can connect a card.
I got the board on taobao(AliExpress China version),the shop called '反客科技'.

THSLP13, 2021-04-20:
**Any one who can leave a comment?I want to know how the issue will be dealed with.**

Because I can't use H743 settings to fit in H750.

dpgeorge, 2021-04-20:
> I can't write a bootloader to jump to app

The stm32 port has a bootloader in `ports/stm32/mboot`.  This bootloader does work on H7 MCUs.  So it's a matter of building a configuration for a H750.

dave2, 2021-05-03:
The H750 (and several other parts like it) are designed to have only sufficient internal flash for a bootloader and some configuration, where the bootloader has to do something more involved to set up further execution. This is discussed in AN5188 with various modes offered.

I don't believe mboot currently supports any of the methods suggested in AN5188 directly. 

AN5188: https://www.st.com/resource/en/application_note/dm00514974-external-memory-code-execution-on-stm32f7x0-value-line-stm32h750-value-line-stm32h7b0-value-line-and-stm32h730-value-line-mcus-stmicroelectronics.pdf