micropython/micropython #7128: CC3200 LAUNCHXL port upip module not found

fsam6177, 2021-04-17:
I installed micropython on CC3200 LAUNCHXL and it works fine. But when I try to import the upip module, it says module not found. I copied the upip.py and upip_utarfile.py to /flash/lib and tried importing the module. It imports the module but cannot find the attributes help() and install() (AttributeError). I also tried copying the files to /flash/src and then deleting the files from /flash/lib and I get the same error. Finally I copied the files to /flash and still no luck. 
Is there a procedure to install the library or get upip to work? 
[image]

dpgeorge, 2021-04-19:
> But when I try to import the upip module, it says module not found.

Can you please show the exact error that it raises?  It may be that it cannot find a dependency of upip, rather than upip itself.  Also please show the output of `sys.path`.

fsam6177, 2021-04-19:
This is what I get when I try to import upip. Also, the sys.path is not available, it is sys.__path__.
[image]

jonnor, 2024-08-18:
`upip` has been replaced by `mip`. Please test and report if `mip` is missing, along with instructions how to reproduce. Proposing to close this issue.

closed 2024-08-18 by robert-hh