micropython/micropython #7129: Pico: sdcard.py timeout waiting for response

jbeale1, 2021-04-18:
I was unable to use the https://github.com/micropython/micropython/blob/master/drivers/sdcard/sdcard.py code as-is on my Raspberry Pi Pico, I would get "timeout waiting for response" when accessing the device, I presume from this part of the code:
```
        # read until start byte (0xff)
        for i in range(_CMD_TIMEOUT):
            self.spi.readinto(self.tokenbuf, 0xFF)
            if self.tokenbuf[0] == _TOKEN_DATA:
                break
        else:
            self.cs(1)
            raise OSError("timeout waiting for response")
```
I was able to fix this by simply changing the _CMD_TIMEOUT from 100 to 1000:
`_CMD_TIMEOUT = const(1000)`
I don't know if the Pico board is faster, or my old SanDisk 8GB is slower than what was tested, but there is an existence proof that more cards will work with a longer timeout. Also, another person using other SanDisk uSD cards reported the same problem, and found the same fix worked for them as well: https://www.raspberrypi.org/forums/viewtopic.php?f=146&t=308819&start=25

dpgeorge, 2021-04-19:
> I was able to fix this by simply changing the _CMD_TIMEOUT from 100 to 1000:

Are you able to test to see what is the minimum timeout that works, eg does 200 work?  If it's just a matter of increasing the timeout then I think that's an OK think to do (it shouldn't break any existing use of this driver).

peterhinch, 2021-04-19:
If revising **sdcard.py** perhaps you might consider https://github.com/micropython/micropython/pull/6007?

jbeale1, 2021-04-21:
In my SD card experiments with the Raspberry Pi Pico writing to a SanDisk 8GB card, I found that the sdcard.py  **_CMD_TIMEOUT** value of 150 worked, but 125 failed.
When it fails, the command, my user-level code that fails is **os.mount(sd, "/SD")**
When it works, that command returns after only 20 msec.  So being more generous with the timeout (eg. 200 or larger) is not going to be a big time-waster in the larger scheme of things. For what it's worth, the SD card spec* requires a host timeout of one (1) full second before failing when issuing the 'ACMD41' command. 
* _SD Specifications Part 1 Physical Layer Simplified Specification Version 8.00_ September 23, 2020, (C) 2001-2020 SD Card Association

dpgeorge, 2021-04-21:
Thanks for testing.  To make it more robust for a wider range of boards (which can run MicroPython at different speeds) I'd suggest adding a delay to this loop, something like:
```diff
--- a/drivers/sdcard/sdcard.py
+++ b/drivers/sdcard/sdcard.py
@@ -176,6 +176,7 @@ class SDCard:
             self.spi.readinto(self.tokenbuf, 0xFF)
             if self.tokenbuf[0] == _TOKEN_DATA:
                 break
+            time.sleep_ms(1)
         else:
             self.cs(1)
             raise OSError("timeout waiting for response")
```
@jbeale1 could you please test that change (without changing the value of `_CMD_TIMEOUT`)?

jbeale1, 2021-04-21:
I have confirmed that the suggested change (adding one line **time.sleep_ms(1)** as indicated) does allow the original timeout value of 100 to work with my SD card.  I agree using a timeout measured in real time units is a better approach, given that different hardware that may run micropython code at different speeds.

dpgeorge, 2021-04-21:
> I have confirmed that the suggested change (adding one line **time.sleep_ms(1)** as indicated) does allow the original timeout value of 100 to work with my SD card.

Great.  So I suggest we go with that change.

dpgeorge, 2021-04-23:
The delay was implemented in a1bc32d8a8fbb09bc04c2ca07b10475f7ddde8c3

pixelpop13, 2022-09-08:
Since I have been unable to find a source for sdcard.py in any github repository (they all seem to be 404, including Adafruit's link),
can someone either post this version of sdcard.py in its entirety or indicate where the sleep_ms(1) line should be placed in the code? I was able to find the code for sdcard.py in an Instructables column.

peterhinch, 2022-09-08:
It has moved [here](https://github.com/micropython/micropython-lib/blob/master/micropython/drivers/storage/sdcard/sdcard.py).

pixelpop13, 2022-09-08:
I am still getting the dreaded "timeout waiting for v2 card". My code appears correct, as it can determine if a card is present or not and its version. I am using a Samsung EVO 32GB card, so I am wondering if this card exceeds the max capacity of the code or if a much longer timeout would allow it to mount properly.

sgbaird, 2022-11-03:
@pixelpop13 I tried it with [the latest on master](https://github.com/micropython/micropython-lib/blob/master/micropython/drivers/storage/sdcard/sdcard.py) and the Samsung 32 GB EVO Plus which came formatted as FAT32, and unfortunately am running into the same issue `timeout waiting for v2 card` https://github.com/micropython/micropython/issues/7051#issuecomment-1301633821. I tried using a 16 GB micro SD card that I had flashed some RPi 400 firmware onto and it worked fine:

Code based on [DigiKey tutorial](https://www.digikey.com/en/maker/projects/raspberry-pi-pico-rp2040-sd-card-example-with-micropython-and-cc/e472c7f578734bfd96d437e68e670050), [adapted for Maker Pi Pico base](https://github.com/CytronTechnologies/MAKER-PI-PICO/issues/4):
```python
import machine
from sdcard import sdcard
import uos

cs = machine.Pin(15, machine.Pin.OUT)

spi = machine.SPI(
    1,
    baudrate=1000000,
    polarity=0,
    phase=0,
    bits=8,
    firstbit=machine.SPI.MSB,
    sck=machine.Pin(10),
    mosi=machine.Pin(11),
    miso=machine.Pin(12),
)

# Initialize SD card
sd = sdcard.SDCard(spi, cs)

vfs = uos.VfsFat(sd)
uos.mount(vfs, "/sd")

with open("/sd/test01.txt", "w") as file:
    file.write("Hello, SD World!\r\n")
    file.write("This is a test\r\n")

# Open the file we just created and read from it
with open("/sd/test01.txt", "r") as file:
    data = file.read()
    print(data)
```
```python
>>> %Run -c $EDITOR_CONTENT
Hello, SD World!
This is a test
```

sgbaird, 2022-11-03:
Hmm, maybe there's a fix that can be implemented for the Samsung EVO 32 GB cards similar to [what was done for CircuitPython](https://github.com/adafruit/Adafruit_CircuitPython_SD/pull/20):
From https://github.com/adafruit/Adafruit_CircuitPython_SD/issues/13:
> The issue is that Samsung EVO 32GB cards require all command codes to have a valid CRC. I used [hazelnusse/crc7@master/crc7.cc](https://github.com/hazelnusse/crc7/blob/master/crc7.cc?rgh-link-date=2019-08-05T18%3A53%3A42Z) to calculate all CRC codes and it is now working. On line 195 I also changed one command from a _block_cmd to a _cmd. It would hang otherwise. This works for my SanDisk 16 GB card which worked before the fix as well and also works on my Samsung EVO 32GB. Please test this with all cards. The spec does say that by default all command code CRC values are ignored except CM0, CMD8 and I believe ACMD41.. It's just that Samsung decided to ignore the default and check all command CRC values.. I was not even able to force it off. Please test the heck out of this. My first fix!
- https://github.com/adafruit/Adafruit_CircuitPython_SD/pull/20

peterhinch, 2022-11-03:
It looks to be SD card specific, not Pico specific. I and Olaf Mastenbroek (@OlafM2015) have done testing to prove the [VS1053 driver](https://github.com/peterhinch/micropython-vs1053) with the Pico. The driver uses the `sdcard` module for its data source. We used a variety of SD cards from 8-32MiB and I tried both SPI buses. We encountered no SD card issues.

pixelpop13, 2022-11-03:
> It looks to be SD card specific, not Pico specific. I and Olaf Mastenbroek (@OlafM2015) have done testing to prove the [VS1053 driver](https://github.com/peterhinch/micropython-vs1053) with the [...]

Thanks! I solved my issue by dropping to an 8GB card, but I will take this under advisement.

CptAmmogeddon, 2023-11-06:
Is there maybe a list with a number of (micro) SD-cards that work? That would be amazing

OlafM2015, 2023-11-06:
Hi,

After several test, my conclusions was Sandisk up to 32 Gb are ok.

Regards,

Olaf

CptAmmogeddon, 2023-11-06:
> Hi, After several test, my conclusions was Sandisk up to 32 Gb are ok.
> […](#)

Thanks! Does it matter if it's SD or SDHC

robert-hh, 2023-11-06:
There is a PR for an update of the driver which aims to work better. You could try that. See the discussion at https://github.com/orgs/micropython/discussions/12687 and his actual version at https://github.com/mendenm/micropython-lib/tree/master/micropython/drivers/storage/sdcard.

robert-hh, 2023-11-06:
> Thanks! Does it matter if it's SD or SDHC

It seems I only use Sandisk cards, both HC and XC. And they work well.

CptAmmogeddon, 2023-11-06:
Sadly, it does not seem to like my Transcend 4GB micro SDHC, even after reformatting.: 
MPY: soft reboot
crc command result: 00
Traceback (most recent call last):
  File "<stdin>", line 23, in <module>
  File "sdcard.py", line 297, in readblocks
  File "sdcard.py", line 226, in readinto
OSError: timeout waiting for response

nkrba, 2025-02-01:
> It looks to be SD card specific, not Pico specific. I and Olaf Mastenbroek ([@OlafM2015](https://github.com/OlafM2015)) have done testing to prove the [VS1053 [...]

Just to note that the `sdcard.py` at [VS1053 driver](https://github.com/peterhinch/micropython-vs1053) uses `_CMD_TIMEOUT = const(1000)` among a few other notable changes from the latest [sdcard.py](https://github.com/micropython/micropython-lib/blob/master/micropython/drivers/storage/sdcard/sdcard.py) as on 1 Feb 2025. Some non-trivial changes from the diff are
```
@@ -100,11 +99,9 @@ class SDCard:
         if csd[0] & 0xC0 == 0x40:  # CSD version 2.0
             self.sectors = ((csd[8] << 8 | csd[9]) + 1) * 1024
         elif csd[0] & 0xC0 == 0x00:  # CSD version 1.0 (old, <=2GB)
-            c_size = (csd[6] & 0b11) << 10 | csd[7] << 2 | csd[8] >> 6
-            c_size_mult = (csd[9] & 0b11) << 1 | csd[10] >> 7
-            read_bl_len = csd[5] & 0b1111
-            capacity = (c_size + 1) * (2 ** (c_size_mult + 2)) * (2**read_bl_len)
-            self.sectors = capacity // 512
+            c_size = csd[6] & 0b11 | csd[7] << 2 | (csd[8] & 0b11000000) << 4
+            c_size_mult = ((csd[9] & 0b11) << 1) | csd[10] >> 7
+            self.sectors = (c_size + 1) * (2 ** (c_size_mult + 2))
         else:
             raise OSError("SD card CSD format not supported")
         # print('sectors', self.sectors)
@@ -114,14 +111,12 @@ class SDCard:
             raise OSError("can't set 512 block size")
 
         # set to high data rate now that it's initialised
-        self.init_spi(baudrate)
+        self.init_spi(1320000)
 
     def init_card_v1(self):
         for i in range(_CMD_TIMEOUT):
-            time.sleep_ms(50)
             self.cmd(55, 0, 0)
             if self.cmd(41, 0, 0) == 0:
-                # SDSC card, uses byte addressing in read/write/erase commands
                 self.cdv = 512
                 # print("[SDCard] v1 card")
                 return
@@ -133,14 +128,8 @@ class SDCard:
             self.cmd(58, 0, 0, 4)
             self.cmd(55, 0, 0)
             if self.cmd(41, 0x40000000, 0) == 0:
-                self.cmd(58, 0, 0, -4)  # 4-byte response, negative means keep the first byte
-                ocr = self.tokenbuf[0]  # get first byte of response, which is OCR
-                if not ocr & 0x40:
-                    # SDSC card, uses byte addressing in read/write/erase commands
-                    self.cdv = 512
-                else:
-                    # SDHC/SDXC card, uses block addressing in read/write/erase commands
-                    self.cdv = 1
+                self.cmd(58, 0, 0, 4)
+                self.cdv = 1
                 # print("[SDCard] v2 card")
                 return
         raise OSError("timeout waiting for v2 card")
@@ -167,10 +156,6 @@ class SDCard:
             response = self.tokenbuf[0]
             if not (response & 0x80):
                 # this could be a big-endian integer that we are getting here
-                # if final<0 then store the first byte to tokenbuf and discard the rest
-                if final < 0:
-                    self.spi.readinto(self.tokenbuf, 0xFF)
-                    final = -1 - final
                 for j in range(final):
                     self.spi.write(b"\xff")
                 if release:
@@ -191,7 +176,6 @@ class SDCard:
             self.spi.readinto(self.tokenbuf, 0xFF)
             if self.tokenbuf[0] == _TOKEN_DATA:
                 break
-            time.sleep_ms(1)
         else:
             self.cs(1)
             raise OSError("timeout waiting for response")
```

peterhinch, 2025-02-02:
> Just to note that the sdcard.py at [VS1053 driver](https://github.com/peterhinch/micropython-vs1053) uses _CMD_TIMEOUT = const(1000)

Indeed, I have found SD cards to vary significantly. Variability is too great to propose a PR as it's not possible to create a reliable repro so my version has a few _ad hoc_ adaptations. This  is one of these cases where expert knowledge is required.