micropython/micropython #7130: ESP32 Bluetooth: Unclear error code "OSError: -18"  when  advertising with a too long payload

LeyangGuo, 2021-04-19:
`MicroPython v1.15 on 2021-04-18; ESP32 module with ESP32`

When trying to advertising with a long payload(longer than 31 bytes) using bluetooth lib on esp32, Micropython would give an unclear error code " OSError: -18 ".
```python
import bluetooth
esp32_ble = bluetooth.BLE()
esp32_ble.active(True)

payload = b'\x02\x01\x06\n\taLongName\x11\x07\x9e\xca\xdc$\x0e\xe5\xa9\xe0\x93\xf3\xa3\xb5\x01\x00@n'

esp32_ble.gap_advertise(interval_us=200000, adv_data=payload)
# Traceback (most recent call last):
#  File "<stdin>", line 6, in <module>
# OSError: -18
```

A payload longer than 31 bytes might be generated with a 128bit Service UUID and a long name(len>8).
```python
from ble_advertising import advertising_payload    #example in /micropython/example/bluetooth

ServiceUUID = bluetooth.UUID("6E400001-B5A3-F393-E0A9-E50E24DCCA9E")
payload = advertising_payload(name="muchTooLong", services=[ServiceUUID])
```
A short name would work just fine.
The error code is not helpful for debugging, but I'm not sure how to improve this.
I opened this issue to notify who have met the same problem.

dpgeorge, 2021-04-19:
I think `OSError(-18)` in this case is referring to the Bluetooth (as in BT spec) error code 0x12, "invalid HCI command parameters".

jimmo, 2021-04-20:
> I think `OSError(-18)` in this case is referring to the Bluetooth (as in BT spec) error code 0x12, "invalid HCI command parameters".

You're right that the 18 comes from 18=`BLE_ERR_INV_HCI_CMD_PARMS`.

However, in this case you'd interpret the return value from `ble_gap_adv_set_data` to be one of the `BLE_HS_*` codes, which in this case a caller should be interpreting it as 18=`BLE_HS_EROLE`.

In general though, NimBLE has a way of passing HCI errors directly through -- it should be 512+18  (i.e. `BLE_HS_HCI_ERR(BLE_ERR_INV_HCI_CMD_PARMS)` ).

I guess there are three workarounds:
 - Fix this in NimBLE (i.e. change https://github.com/apache/mynewt-nimble/blob/master/nimble/host/src/ble_gap.c#L2564 and many other places to use the `BLE_HS_HCI_ERR` macro).
 - Alternatively, fix in NimBLE by having it return `BLE_HS_EINVAL` instead (and modbluetooth_nimble would already convert that to `OSError(EINVAL)`).
 - Workaround this in modbluetooth_nimble.c (i.e. detect the non-nonsensical `BLE_HS_EROLE` and map it to EINVAL)
 - Do the length check in modbluetooth.c instead

dpgeorge, 2021-04-20:
> In general though, NimBLE has a way of passing HCI errors directly through -- it should be 512+18

Yes you're right.  But esp32 does use a modified NimBLE so maybe they did something else here... I guess it's easy to check this behaviour on a PYBD.

> Fix this in NimBLE

If NimBLE really are doing this wrong then IMO this is the right fix, although it'll probably take a while to filter down here.

sergioro99, 2022-07-02:
Hello, I have exactly the same problem and I don't know how to solve it. 
What can I do? 
Thank you very much for your help :)

jimmo, 2022-07-04:
> EDIT: I solved it by setting a bluetooth network name shorter (as LeyangGuo said) so the payload is no longer than 31 bytes and it works. It works, however I don't see it as the best solution.

@sergioro99 This is a limitation of the way advertising works. You cannot create a payload longer than 31 bytes, so this needs to be fixed in your application (e.g. by reducing the name length). The issue in question here is how the error is reported.

The one option you have is to also use the scan response for additional fields (i.e. put your service uuid in the adv payload, and name in the scan response). In [aioble](https://github.com/micropython/micropython-lib/tree/master/micropython/bluetooth/aioble) we automatically spread the fields across both the advertising payload and scan response as necessary.

lacmuch, 2023-02-28:
Some info about the adv_data format:

https://docs.silabs.com/bluetooth/4.0/general/adv-and-scanning/bluetooth-adv-data-basics