micropython/micropython #7134: ESP32 I2C: Repeated start handling incorrect

jetpax, 2021-04-20:
The I2C 'repeated start' handling code for ESP32 fails to drive the ESP-IDF correctly, which leads to very inefficient blocking use of the bus when when register/memory reads are performed (of course, this is only of concern if high I2C performance on a busy bus is required).

For example, see the large dead time between register setup and eventual read in the trace below.

[image]

 
This is caused by an elegant, but in the end insufficient  `read_mem`  function in `ports/esp32/machine_i2c.c`.

By modifying `read_mem` to use an updated `machine_hw_i2c_transfer`  in  `ports/esp32/machine_i2c.c`, the problem is resolved:

[image]

This also makes `read_mem` consistent  with `write_mem`.

This has been tested with 1MHz bus operation using  a wide variety of sensors and addressing modes; happy to raise a PR if there is interest (ie if it is ever likely to be merged)

pumelo, 2021-04-24:
I can't talk for the maintainers but I am personally interested in the solution to this issue. Would you mind sharing it anyway?

jetpax, 2021-04-25:
The changes are very simple

in  `ports/esp32/machine_i2c.c` add the following to `machine_hw_i2c_transfer` 

```   
    int data_len = 0;
    if (n > 1) {
        // >1 buffers => first cycle must be write so Q slave addr (W)
        i2c_master_write_byte(cmd, addr << 1, true);
        // Q first buffer
        i2c_master_write(cmd, bufs->buf, bufs->len, true);
        data_len += bufs->len;
        ++bufs;
        --n;
        if (flags & MP_MACHINE_I2C_FLAG_READ) {
            // this is a register read so Q repeated start with slave addr (R)
            i2c_master_start(cmd);
            i2c_master_write_byte(cmd, addr << 1 | MP_MACHINE_I2C_FLAG_READ, true);
        }
    } else {
        // 0/1 buffer => simple transaction, so Q slave address with (R/W) as passed
        i2c_master_write_byte(cmd, addr << 1 | (flags & MP_MACHINE_I2C_FLAG_READ), true);
    }
```
and change the `read_mem` calling function in `extmod/machinei2c.c` to use `machine_hw_i2c_transfer` (same as `write_mem`)

```
STATIC int read_mem(mp_obj_t self_in, uint16_t addr, uint32_t memaddr, uint8_t addrsize, uint8_t *buf, size_t len) {
    mp_obj_base_t *self = (mp_obj_base_t *)MP_OBJ_TO_PTR(self_in);
    // Create buffer with memory address
    uint8_t memaddr_buf[4];
    size_t memaddr_len = fill_memaddr_buf(&memaddr_buf[0], memaddr, addrsize);
    // Create partial write buffers
    mp_machine_i2c_buf_t bufs[2] = {
        {.len = memaddr_len, .buf = memaddr_buf},
        {.len = len, .buf = (uint8_t *)buf},
    };
    // Do I2C transfer
    mp_machine_i2c_p_t *i2c_p = (mp_machine_i2c_p_t *)self->type->protocol;
    unsigned int flags = MP_MACHINE_I2C_FLAG_READ | MP_MACHINE_I2C_FLAG_STOP;
    return i2c_p->transfer(self, addr, 2, bufs, flags);
}
```

These changes are in my [fork](https://github.com/jetpax/jep-micropython.git)

3ym3ym, 2021-05-13:
See this exact issue as well. My workaround is to use SoftI2C rather than I2C class.
Maintainers, please address.

ozel, 2022-05-03:
any reason why this fix hasn't been incorporated, yet?

dpgeorge, 2022-06-02:
Commit 6bda80d81147217a1d830b99b93d2e35d372e8f9 improves the efficiency of I2C transfers by doing the write and read in one call to the IDF.

closed 2022-06-02 by dpgeorge