micropython/micropython #7139: WLAN.connect() behavior has changed

the-stanely, 2021-04-20:
This problem is specific to `esp32-20210418-v1.15.bin`.

When preparing WiFi like,

```
station = network.WLAN(network.STA_IF)
station.active(True)
```

And then connecting with,

`station.connect("mynet", "mypass")`

I used to be able to attempt to reconnect by repeating the above statements, regardless if the station is already connected.  In this new version, the third -- not second, but third -- time I say,

`station.connect("mynet", "mypass")`

If the station is already connected, I get this error,

> Traceback (most recent call last):
>  File "<stdin>", line 1, in \<module\>
> OSError: Wifi Internal Error

This can be easily tested from REPL.

The workaround for this is to test for a connection, disconnect, and reconnect as in,

```
if station.isconnected():
    station.disconnect()
station.connect("mynet", "mypass")
```

If this was intentional, it will likely break a lot of code -- well... at least a lot of my code.  :)

But it it's intentional, why error on 3rd and not 2nd time?

dpgeorge, 2021-04-21:
Thanks for the report, I can reproduce the issue.  My guess would be the change is due to using IDF v4.

In your previous code, do you know what version of the IDF you used, and what version of MicroPython worked?

the-stanely, 2021-04-21:
I used both idf4 and idf3.  In fact I tried both in various versions trying to get mqtt ssl/tls to work.  I found that only idf3 v1.12 worked with AWS IoT, but was overjoyed to see that AWS also works in 1.15! Then I discovered this strange WiFi connect behavior.

v1.15 is the first time I've seen this problem.

fliiiix, 2022-08-28:
I seen the same issue on v1.18 (2022-01-17) so i guess there is no fix?
But not on v1.17 (2021-09-02) :eyes:

jonnor, 2024-08-18:
Is this issue still present on MicroPython 1.23 ?

projectgus, 2024-08-27:
I'm going to close this as I think it's an artifact of ESP-IDF behaviour. `WLAN.connect()` starts an internal state machine inside ESP-IDF to connect to Wi-Fi (multiple steps, multiple layers of the network stack involved, etc.) So something like the workaround described of `WLAN.disconnect()` and then `WLAN.connect()` is probably the best way to "reset" this if you need to. Although a lot of the time it's probably unnecessary and it may be better to monitor `WLAN.status()` to see if a connect is still in progress or has failed for a known reason.

closed 2024-08-27 by projectgus