micropython/micropython #7144: round(1,1) will raise NotImplementedError in windows port

shaoziyang, 2021-04-21:
round(1,1) will raise NotImplementedError in windows port. It woks in other port.

```
>>> round(1,1)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NotImplementedError:
>>> round(1.0,1)
1.0
```

dpgeorge, 2021-04-22:
This was enabled on the unix port in 20b4b85f7266ef2edbe492829585a14118a85d7a, so probably makes sense to enable it on Windows as well.  In fact, could also enable the following to make Windows closer to unix in configuration:
```c
#define MICROPY_PY_DELATTR_SETATTR  (1)
#define MICROPY_PY_BUILTINS_ROUND_INT    (1)
#define MICROPY_PY_BUILTINS_SLICE_INDICES (1)
#define MICROPY_PY_IO_IOBASE        (1)
```

stinos, 2021-04-22:
I'll do another round of 'make windows ports similar to unix port' PR, has been a while.

@shaoziyang do you use mingw or msvc? In the first case you can build with the standard `make CFLAGS_EXTRA="-DMICROPY_PY_BUILTINS_ROUND_INT=1"`, in the latter case there are the standard msbuild extension points like Directory.build.props etc and one specific to MicroPython as well: you can create a `user.props` file in the windows/msvc directory with 
```
<?xml version="1.0" encoding="utf-8"?>
<Project ToolsVersion="4.0" xmlns="http://schemas.microsoft.com/developer/msbuild/2003">
  <ItemDefinitionGroup>
    <ClCompile>
      <PreprocessorDefinitions>MICROPY_PY_BUILTINS_ROUND_INT=1;%(PreprocessorDefinitions)</PreprocessorDefinitions>
    </ClCompile>
  </ItemDefinitionGroup>
</Project>
```

dpgeorge, 2021-04-23:
Fixed by bb2007b05ccf0e41684cc61f1677c11b7df93f7a

closed 2021-04-23 by dpgeorge