micropython/micropython #7148: STM32: How to avoid disabling USB IRQs in sdcard.c ?

iabdalkader, 2021-04-22:
https://github.com/micropython/micropython/blob/00d6a79b3d5dc80d840dc1d51166e7d95856b3d6/ports/stm32/sdcard.c#L599

Any suggestions for a better way to handle this other than disabling USB IRQs ? This is starting to be an issue especially when reading/writing many blocks in one call or reading/writing frequently in a loop, all USB IRQs are disabled and this stops other things that don't access MSC at all from working.

I tried setting a flag in sdcard.c when reading/writing instead of disabling IRQs, to indicate the card is busy and checking this flag in `usbcd_msc_interface.c` returning -1 from `usbd_msc_IsReady()` if the flag is set, it almost works except on the host I see I/O errors and things eventually fail... Is there anything else I can try ? 

```diff
diff --git a/ports/stm32/usbd_msc_interface.c b/ports/stm32/usbd_msc_interface.c
index 62fe32bf5..71eb58259 100644
--- a/ports/stm32/usbd_msc_interface.c
+++ b/ports/stm32/usbd_msc_interface.c
@@ -242,9 +242,10 @@ STATIC int8_t usbd_msc_GetCapacity(uint8_t lun, uint32_t *block_num, uint16_t *b
     return lu_ioctl(lun, MP_BLOCKDEV_IOCTL_BLOCK_COUNT, block_num);
 }
 
+extern volatile uint32_t sdcard_busy;
 // Check if a logical unit is ready
 STATIC int8_t usbd_msc_IsReady(uint8_t lun) {
-    if (lun >= usbd_msc_lu_num) {
+    if (lun >= usbd_msc_lu_num || sdcard_busy) {
         return -1;
     }
     return lu_flag_is_set(lun, FLAGS_STARTED) ? 0 : -1;
``` 

Also tried returning different status codes from here: https://publib.boulder.ibm.com/tividd/td/TSMM/GC32-0767-00/en_US/HTML/anrcms45.htm but all of them cause I/O errors on the host.

dpgeorge, 2021-04-23:
Is it a solution to simply disable USB MSC, ie only allow USB VCP and/or HID?

iabdalkader, 2021-04-23:
> Is it a solution to simply disable USB MSC, ie only allow USB VCP and/or HID?

@dpgeorge That still wouldn't fix it because sdcard_read/write_blocks disables USB IRQs regardless of whether MSC is enabled or not (which is exactly why this is an issue), but other than that I still need MSC in normal cases...After some more testing I found out that returning SCSI ASC 0x04 (Logical unit not ready) is much more stable than 0x3A (Not present). When 0x3A is returned (the default) the disk keeps disappearing, but with 0x04 instead I just get an error when accessing the disk if the SD card is too busy, and it works normally if it's not, all while keeping USB IRQs enabled. The "error" messages look like this:
```bash
[ 9869.604749] blk_update_request: I/O error, dev sda, sector 284256 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[ 9869.604754] Buffer I/O error on dev sda, logical block 35532, async page read
[ 9900.301021] sd 0:0:0:0: [sda] tag#0 FAILED Result: hostbyte=DID_OK driverbyte=DRIVER_SENSE cmd_age=0s
[ 9900.301031] sd 0:0:0:0: [sda] tag#0 Sense Key : Not Ready [current] 
[ 9900.301034] sd 0:0:0:0: [sda] tag#0 Add. Sense: Logical unit not ready, cause not reportable
[ 9900.301038] sd 0:0:0:0: [sda] tag#0 CDB: Write(10) 2a 00 00 00 12 00 00 00 08 00
```
 I'm not sure if all OSes handle the SCSI codes the same way, we'll have to test it to find out but it looks like a promising solution.

dpgeorge, 2021-04-23:
> That still wouldn't fix it because sdcard_read/write_blocks disables USB IRQs regardless of whether MSC is enabled or not

I understand.  So one improvement would be to not disable USB IRQs when MSC is not being used (ie not enabled or currently not mounted by the host/PC).

> I found out that returning SCSI ASC 0x04 (Logical unit not ready) is much more stable than 0x3A (Not present)

Nice!  If that works across different OS's would be good to implement (does TinyUSB use this approach for its MSC?).

iabdalkader, 2021-04-24:
>  So one improvement would be to not disable USB IRQs when MSC is not being used (ie not enabled or currently not mounted by the host/PC).

Yes that's a good first step, I can send a PR but how can I detect that in sdcard.c ?

> Nice! If that works across different OS's would be good to implement (does TinyUSB use this approach for its MSC?).

Unfortunately my approach has issues, for example if the MSC finds the card busy even once it will return not ready/present, and if the host was accessing the device it will issue a reset to the USB device. I was thinking maybe it's best to just disable MSC if the SD is accessed from Python even once and then it gets re-enabled on softreset ? The host side and the device side can't see new changes anyway.

dpgeorge, 2021-04-29:
I have an idea how to make this work: try to decouple the low-level USB interrupt code from the SCSI driver, and run the SCSI driver only when the storage (eg SD card) is not being used.

iabdalkader, 2021-04-30:
> I have an idea how to make this work: try to decouple the low-level USB interrupt code from the SCSI driver, and run the SCSI driver only when the storage (eg SD card) is not being used.

Do you mean like in PendSV context ?

dpgeorge, 2021-04-30:
> Do you mean like in PendSV context ?

Yes, or some other way to delay processing the SCSI request.  In the current code, as soon as it gets a SCSI request it responds.  I think the response can be delayed, so the idea would be to "set a flag" in the USB IRQ handler that there is an outstanding SCSI request, and then handle this at the next most convenient time (eg via `mp_sched_schedule()`, or `pendsv_schedule_dispatch()`).

Might be worth checking if/how TinyUSB handles this.