micropython/micropython #7149: docs: document uwebsocket module

dpgeorge, 2021-04-23:
It needs basic docs to explain that it's mainly used for `webrepl` and does not have a full websocket implementation.

See related: #2410, #4450.

rubydesign, 2021-04-23:
I would be happy to proof read any such docs, or provide an end user (newbie) perspective.

I was researching websockets, and i think my basic use case of building a home grown communication between two or more esp modules with a websocket client and the async websocket server is quite common.
In other words even non standard, non complete solution has many.
I would also be happy to write a client/server example once basic docs are there.

etienne07C1, 2023-04-29:
Currently assessing the possibility of migrating to MicroPython, just wondered about the state of websocket in upython: is it enough mature to provide same features than C++ implementation?

rosskusler, 2024-03-18:
I'd love to see websocket support in micropython too. It would be fantastic for some IOT use cases