micropython/micropython #7150: rp2: add support for SPARKFUN THING PLUS - RP2040 (16MB flash)

jgriesfeller, 2021-04-23:
Hi, 

I just got some [SPARKFUN THING PLUS - RP2040](https://www.sparkfun.com/products/17745url) boards with 16MB flash. It would be nice to have that flash supported somehow. The standard pico build works otherwise.

I'm not deep enough into Micropython to do that myself, but I am happy to test if someone starts the work.

robert-hh, 2021-04-23:
Try creating a board definition along the files in rp2/boards, e.g. rp2/boards/ADAFRUIT_FEATHER_RP2040

robert-hh, 2021-04-27:
If you did not make it yet, youj can extact the attached zip to the rp2/boards directory and run:

make BOARD=SPARKFUN_THINGPLUS

which will create a build_SPARKFUN_THINGPLUS directory with the .uf2 file in it.

[SPARKFUN_THINGPLUS.zip](https://github.com/micropython/micropython/files/6383750/SPARKFUN_THINGPLUS.zip)

jgriesfeller, 2021-04-28:
@robert-hh Thanks for making the effort for me!

In principle I wanted to provide a pull request for this, but unfortunately the build process for rp20240  seems to be much less streamlined compared to the other ports I have compiled so far (ESP32 and stm32). 
Following the instructions I am not able to build (but I might have missed something):
```
(micropython) jang@pc5378:~/.../ports/rp2$ git pull
Already up to date.
(micropython) jang@pc5378:~/.../ports/rp2$ make clean
rm -f -rf build-PICO
(micropython) jang@pc5378:~/.../ports/rp2$ make
[ -d build-PICO ] || cmake -S . -B build-PICO -DPICO_BUILD_DOCS=0 -DMICROPY_BOARD=PICO
CMake Error: The source directory "/home/jang/data/Python3/micropython/ports/rp2/build-PICO" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.
Makefile:18: recipe for target 'all' failed
make: *** [all] Error 1
```
creating the build directory by hand does not help either:
```
(micropython) jang@pc5378:~/.../ports/rp2$ mkdir build-PICO
(micropython) jang@pc5378:~/.../ports/rp2$ make
[ -d build-PICO ] || cmake -S . -B build-PICO -DPICO_BUILD_DOCS=0 -DMICROPY_BOARD=PICO
make -s -C build-PICO
make[1]: *** No targets specified and no makefile found.  Stop.
Makefile:18: recipe for target 'all' failed
make: *** [all] Error 2
(micropython) jang@pc5378:~/.../ports/rp2$ make board=PICO
[ -d build-PICO ] || cmake -S . -B build-PICO -DPICO_BUILD_DOCS=0 -DMICROPY_BOARD=PICO
make -s -C build-PICO
make[1]: *** No targets specified and no makefile found.  Stop.
Makefile:18: recipe for target 'all' failed
make: *** [all] Error 2
```
I'll try to fix this myself, but i might have another issue to report...

robert-hh, 2021-04-28:
That looks okay, but the build does not start.
Did you update the submodules?
Which version of cmake do you use? AFAIK it should be 3.12 at least?
Does the rp2 directory contain the following files and directories?
```
boards                   machine_pin.c    Makefile      modules         mpthreadport.h  tusb_port.c
build-PICO               machine_pwm.c    manifest.py   moduos.c        qstrdefsport.h  uart.c
build-SPARKFUN_PROMICRO  machine_spi.c    memmap_mp.ld  modutime.c      README.md       uart.h
CMakeLists.txt           machine_timer.c  modmachine.c  mpconfigport.h  rp2_flash.c
fatfs_port.c             machine_uart.c   modmachine.h  mphalport.c     rp2_pio.c
machine_adc.c            machine_wdt.c    modrp2.c      mphalport.h 
machine_i2c.c            main.c           modrp2.h      mpthreadport.c  tusb_config.h
```

jgriesfeller, 2021-04-29:
Found several problems:
- cmake version (too old in Ubuntu bionic). Solved via conda
- make did not catch the PICO_SDK_PATH_OVERRIDE environment variable; solved with a symbolic link
- I was missing the submodules; I ran `git submodule update --recursive` to solve

I can build micropython now. Pull request will come soon.

BTW, I will get an Adafruit ItsyBitsy RP2040 soon as well. Should that also get its own board definition or will you just document somewhere that the Feather board definition should be used? It has the same 8M flash.

robert-hh, 2021-04-29:
You can do either thing. But if, you have do use the board names following the include file names in pico-sdk/src/boards/include/boards. So the board name directory for the Adafruit device would be
`ADAFRUIT_ITSYBITSY_RP2040`

jgriesfeller, 2021-04-29:
Pull request for Sparkfun Thing Plus created (#7186)

jgriesfeller, 2021-05-05:
closing this issue since the Pr has been merged

closed 2021-05-05 by jgriesfeller