micropython/micropython #7151: LinkError: build/modrun.o: undefined symbol: mp_obj_get_array

KejaPower, 2021-04-23:
Hello,
I'm trying to build a .mpy file , the C code of the function that will be called from micropython is: 
```
#include "py/dynruntime.h"

STATIC mp_obj_t run_t( mp_obj_t pylistin) {
  size_t len=0;
  mp_obj_t *features;
  mp_obj_get_array(pylistin,&len, &features);
  mp_float_t cfeatures[len]  ;
for (int i=0; i < len;++i)
{
cfeatures[i]=mp_obj_get_float(features[i]);
}

return mp_obj_new_float(cfeatures[0]);

}
```

I get the following error when compiling : 
```
GEN build/modrun.config.h
CC modrun.c
LINK build/modrun.o
LinkError: build/modrun.o: undefined symbol: mp_obj_get_array
make: *** [build/modrun.native.mpy] Error 1
```
my make file is : 
```

# Location of top-level MicroPython directory
MPY_DIR = ../../..

# Name of module
MOD = modrun

# Source files (.c or .py)
SRC = modrun.c

# Architecture to build for (x86, x64, armv7m, xtensa, xtensawin)
ARCH = armv7emsp
#ARCH = x64

```

# Include to get the rules for compiling and linking the module
include $(MPY_DIR)/py/dynruntime.mk

I have tried the natmod examples and they work, it seems the issue is when I m using mp_obj_get_array.

What am I doing wrong?
thanks

dpgeorge, 2021-04-23:
This function is currently not implemented for dynamic runtime.  But it's easy enough to add, see #7154.

KejaPower, 2021-04-24:
thanks. I managed to get it working but I had those compilation errors. : 
LinkError: build/modrun.o: undefined symbol: mp_raise_TypeError
or
LinkError: build/modrun.o: undefined symbol: mp_type_tuple

dpgeorge, 2021-04-24:
> I managed to get it working but I had those compilation errors.

I think I fixed these now, please try that PR again.

KejaPower, 2021-04-24:
I m still getting this error :  `LinkError: build/modrun.o: undefined symbol: mp_raise_TypeError`

dpgeorge, 2021-04-24:
It compiles for me... can you post the full code (including Makefile) that has the error?

KejaPower, 2021-04-24:
The full code : 
```
#include "py/dynruntime.h"
#include "prod.h"

STATIC mp_obj_t run_classifier_t( mp_obj_t pylistin) {

  size_t len;

  mp_obj_t *features;

  mp_obj_get_array(pylistin,&len, &features);

  mp_float_t cfeatures[len]  ;

for (size_t i=0; i < len;++i)
{
  float t;
cfeatures[i]=mp_obj_get_float(features[i]);
t=prod_array(cfeatures[i]);
cfeatures[i]=t;
features[i]=mp_obj_new_float(cfeatures[i]);

}
mp_obj_t listout=mp_obj_new_list(len,features);

return listout;
}
;
 

STATIC MP_DEFINE_CONST_FUN_OBJ_1(run_classifier_obj, run_classifier_t);

// This is the entry point and is called when the module is imported
mp_obj_t mpy_init(mp_obj_fun_bc_t *self, size_t n_args, size_t n_kw, mp_obj_t *args) {
    // This must be first, it sets up the globals dict and other things
    MP_DYNRUNTIME_INIT_ENTRY

    // Make the functions available in the module's namespace
    mp_store_global(MP_QSTR_run_classifier, MP_OBJ_FROM_PTR(&run_classifier_obj));

    // This must be last, it restores the globals dict
    MP_DYNRUNTIME_INIT_EXIT
}
```

The Makefile : 
```

# Location of top-level MicroPython directory
MPY_DIR = ../../..

# Name of module
MOD = modrun

# Source files (.c or .py)
SRC = modrun.c prod.c

# Architecture to build for (x86, x64, armv7m, xtensa, xtensawin)
ARCH = armv7emsp
#ARCH = x64

# Include to get the rules for compiling and linking the module
include $(MPY_DIR)/py/dynruntime.mk
```

dpgeorge, 2021-04-24:
Compiling that example (and copying prod.c/h from features2) it gives:
```
$ make
CC modrun.c
modrun.c: In function 'run_classifier_t':
modrun.c:19:3: error: too few arguments to function 'prod_array'
   19 | t=prod_array(cfeatures[i]);
      |   ^~~~~~~~~~
In file included from modrun.c:2:
prod.h:1:7: note: declared here
    1 | float prod_array(int n, float *ar);
      |       ^~~~~~~~~~
make: *** [../../../py/dynruntime.mk:130: build/modrun.o] Error 1
```
After fixing that issue, it compiles fine.

dpgeorge, 2021-04-24:
Note that you'll need to re fetch/pull the PR to get the latest changes (the branch was force-pushed).

KejaPower, 2021-04-24:
thank you very much, works for me too

closed 2021-04-24 by KejaPower