micropython/micropython #7152: drivers/nrf24l01/nrf24l01.py missing power up delay.

ip2016, 2021-04-23:
I'm new to micropython + nrf24l01, so when started with my little project I noticed that only every second packet sent by transmitter is received by a receiver. Digging into the issue I found out that when send is timed out the method exits without putting nrf24 into power down mode and then the next send attempt success. So we're getting the following sequence:
power up -> send -> timeout (no data received)-> send -> success -> power down -> power up -> send -> timeout... ->...
Digging even deeper I found that nrf24 doesn't even try to send data after power up and timeouts. 
Now reading the nRF24 documentation and source code of the driver I noticed that delay after power up is only 150us instead of 1.5ms as per documentation  (nRF24L01_Product_Specification_v2_0.pdf - 6.1.7 Timing Information). 
Increasing the delay fixed the issue for me.
Setup: ESP8266 (Wemos D1 mini), nRF24l01+

peterhinch, 2021-04-24:
I assume you're referring to this line https://github.com/micropython/micropython/blob/530c76f6caee9445f468380f58b98f4e3ceda759/drivers/nrf24l01/nrf24l01.py#L230

I must say I'm puzzled. I've used these radios extensively with this driver and have never encountered a problem here. I also have a project which has run continuously for several years where four Pyboards communicate every hour via nrf24l01+ radios. The usual reason for transmission failure is that the receiver is not in listening mode at the time of transmission. Are you sure you've excluded this possibility?

ip2016, 2021-04-24:
Yes, I excluded this possibility. 
First thing I did was removing timeout here:
https://github.com/micropython/micropython/blob/530c76f6caee9445f468380f58b98f4e3ceda759/drivers/nrf24l01/nrf24l01.py#L221
And I got nrf24 stuck at sending forever. Which means no TX_DS  or MAX_RT were raised. Then I returned the timeout and modified the code to print the status of nrf24 after each send attempt. What I got is every second send was timed out and for each timed out send the max retransmit count was zero. Clearly the PTX was not sending a packet due to failed initialization. After timed out send,  the PTX was not going into power down mode, so subsequent send was succeeded.
It might be something to do with my nRF24 modules.
The manual stays:

> When nRF24L01 is in power down mode it must settle for 1.5ms before it can enter the TX or RX modes. If 
> an external clock is used this delay is reduced to 150µs, see Table 13. on page 22. The settling time must 
> be controlled by the MCU.

So it might be your modules are using external oscillator and mine - internal.

peterhinch, 2021-04-24:
You could be right. Mine are Sparkfun.

jonnor, 2024-09-13:
Is this still relevant?

Note that the driver is now at https://github.com/micropython/micropython-lib/tree/27e4d73bc2618d378a0610960cf5e81985e5d914/micropython/drivers/radio/nrf24l01

peterhinch, 2024-09-14:
Closing due to inactivity.

closed 2024-09-14 by peterhinch