micropython/micropython #7158: ESP32 calling sleep function without arguments after calling a sleep function previously with expiry causes sleep to expire at previously set expiry

VynDragon, 2021-04-24:
Hello.

In code, if I do machine.<some>sleep(some_expiry), then later on call machine.<some>sleep(), the ESP32 wakes after the expiry set in the first call with machine.wake_reason() as 4 (TIMER_WAKE, which is undocumented btw).
It seems to be because the wakeup trigger is never disabled as only esp_sleep_enable_timer_wakeup is called and never esp_sleep_disable_timer_wakeup, that i could find.
see https://github.com/espressif/esp-idf/issues/1677

PBrunot, 2022-11-30:
Maybe disabling sources before re-configuring them would be enough ? 
see concept below esp32/modmachine.c , added 2 lines including one comment.

```c
STATIC mp_obj_t machine_sleep_helper(wake_type_t wake_type, size_t n_args, const mp_obj_t *pos_args, mp_map_t *kw_args) {

    enum {ARG_sleep_ms};
    const mp_arg_t allowed_args[] = {
        { MP_QSTR_sleep_ms, MP_ARG_INT, { .u_int = 0 } },
    };

    mp_arg_val_t args[MP_ARRAY_SIZE(allowed_args)];
    mp_arg_parse_all(n_args, pos_args, kw_args, MP_ARRAY_SIZE(allowed_args), allowed_args, args);

    mp_int_t expiry = args[ARG_sleep_ms].u_int;

    // First disable any previously set wake-up sources
    esp_sleep_disable_wakeup_source(ESP_SLEEP_WAKEUP_ALL);

    if (expiry != 0) {
        esp_sleep_enable_timer_wakeup(((uint64_t)expiry) * 1000);
    }
```

mgsb, 2023-10-22:
+1
This affects any wakeup source(s). So, if you use wake_on_ext0(pin) and then later try wake_on_ext0(None), the pin will still be active as a wake source with subsequent lightsleep()
calls.
I think the patch proposed by @PBrunot looks good. You could individually disable sources, but since they are individually checked for (re)enabling, it's probably more efficient to disable all in one call.

PBrunot, 2023-10-22:
Thanks for the feedback. I'll prepare a PR accordingly.

closed 2023-10-31 by dpgeorge