micropython/micropython #7160: Wemos D1 mini boot.py bricked (ESP8266)

nissene99, 2021-04-25:
Since I accidentally have an error in my boot.py, the ESP8266 does not boot anymore. I already tried to reset it via pin D3 -> GND, D8 -> GND or D4 -> 3.3V, but it didn't work. Therefore the ESP is not recognized when connecting to the PC. Neither from Linux nor from Windows.

boot.py:

    #import esp
    #esp.osdebug(None)
    import uos, machine, gc, webrepl
    try:
        import usocket as socket
    except:
        import socket
    try:
        import ustruct as struct
    except:
        import struct
    NTP_DELTA = 3155673600
    host = "pool.ntp.org"
    #uos.dupterm(None, 1) # disable REPL on UART(0)
    webrepl.start()
    gc.collect()
    
    def time():
        NTP_QUERY = bytearray(48)
        NTP_QUERY[0] = 0x1B
        addr = socket.getaddrinfo(host, 123)[0][-1]
        s = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
        try:
            s.settimeout(1)
            res = s.sendto(NTP_QUERY, addr)
            msg = s.recv(48)
        finally:
            s.close()
        val = struct.unpack("!I", msg[40:44])[0]
        return val - NTP_DELTA
    def settime():
        t = time()
        import machine, utime
        tm = utime.localtime(t)
        machine.RTC().datetime((tm[0], tm[1], tm[2], tm[6] + 1, tm[3], tm[4], tm[5], 0))
        print("Date\t",tm[2],".",tm[1],".",tm[0]"\n\nTime:\t", tm[6] + 1,":",tm[3])
    
    def connect_wifi():
        import network
        ap = network.WLAN(network.AP_IF)
        sta = network.WLAN(network.STA_IF)
        ap.active(False)
        if not sta.isconnected():
            sta.active(True)
            sta.connect('#####','#####')
            while not sta.isconnected():
                pass
        else:
            settime()
            pass
        sta.ifconfig(('192.168.188.18', '255.255.255.0', '192.168.188.1', '192.168.188.1'))
    connect_wifi()

Is there a way to clear the memory of the ESP manually or another way to overwrite the firmware?

robert-hh, 2021-04-25:
You can clear the memory using esptool.py and the option erase_flash, or with the Makefile from the repository, calling
make erase
which in turn calls  esptool.py. Then you can start over by uploading the firmware and setting up the file system#s content.
But in any case, the esp's UArt should be detected.
b.t.w.: This is the bug tracker. For usage questions like yours, better ask at the forum: forum.micropython.org.

nissene99, 2021-04-25:
okay thank you for the quick reply. Have now posted the whole thing in the forum.

closed 2021-04-25 by nissene99