micropython/micropython #7165: PyCharm warning when using pin.value()

Delilovic, 2021-04-26:
PyCharm is displaying a warning when using pin.value()

Parameter 'x' undefined

[image]

dpgeorge, 2021-04-27:
This must be a problem with PyCharm set-up, that it doesn't know the correct signature of `Pin.value()`.  How are function signatures configured?

mishafarms, 2021-05-06:
There is a plugin for PyCharm for micropython, You must install it and then enable it for your project. See settings/Plugins to install and settings/Languages & Frameworks/micropython.
It does not have support for esp32 just esp8266, but machine seems to be pretty well supported. The plugin source in at 
[https://github.com/vlasovskikh/intellij-micropython](url)

dpgeorge, 2021-05-06:
Thanks @mishafarms for the info.  I think this issue can be closed now.

closed 2021-05-06 by dpgeorge