micropython/micropython #7166: Event like sleep_ms, print  cause IRQ handler function to freeze on ESP32

ypj0202, 2021-04-26:
I am running in a problem with interrupt in Micropython. If I try to print something or wait by using sleep_ms, micropython simply just freeze, cannot interrupt by anything (Even from REPL). Only reset button can re-run the program again but it will freeze again once interrupt occurred. The only situation not causing it to freeze is changing global variable.
[image]
Above is my interrupt for button (Wait for button to debounce)
[image]
How I declare my button interrupt.

Tested on Micropython 1.14 and 1.15 with no hope.

Please help!

glenn20, 2021-04-27:
There are significant limitations on what you can do in an irq handler. See:
https://docs.micropython.org/en/latest/reference/isr_rules.html

These restrictions are a consequence of the realities of microcontroller hardware (eg. other interrupts masked while ISR executes, memory allocation failing, etc).

One approach which may work for you is to write a short and sweet mid_button_interrupt() function which just calls micropython.schedule() to schedule a second handler to execute "as soon as feasible". That second handler would not be bound by the same restrictions as the first handler. See the docs above for an example.

BTW. This sort of question should be asked in the micropython  forums (https://forum.micropython.org/), rather than raised as a software issue (bugs, feature requests, ..). Good luck.

ypj0202, 2021-04-27:
Thanks for the reply, the test with the similar thing already the other day by using `Timer` and tried the way that you suggested.
[image]
The problem still exist, as I mentioned before it will work if there is only changing the global variables. In the screenshot above, I add a print statement and a wait, which cause it to freeze again(Not crashing) without any prompt in REPL.

Did I do something wrong?

dpgeorge, 2021-04-28:
Please can you post a full (minimal) example which shows the issue when run?  Then we can see if it's a bug or not.

ypj0202, 2021-05-25:
Sorry for the last reply, I try more on my code, it turn out that thread is the main thing that cause the interrupt not working. Before I run the thread, interrupt can work as expected, but after the thread is started, and interrupt is triggered, it will simply cause micropython to freeze. Is that a bug or it is how its designed?

dpgeorge, 2021-05-30:
> Before I run the thread, interrupt can work as expected, but after the thread is started, and interrupt is triggered, it will simply cause micropython to freeze.

Are you able to provide some code which shows the problem?  This feature should work, so it's a bug and to help fix it we need to be able to reproduce it.

closed 2021-06-12 by ypj0202