micropython/micropython #7171: Exceptions in class level mess up list and dict comprehensions afterwards

aivarannamaa, 2021-04-27:
First I submit code, which has errors on class level:

```
MPY: soft reboot
MicroPython v1.15 on 2021-04-18; ESP module with ESP8266
Type "help()" for more information.
>>> class X:
...     blah()
...     
...     
... 
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 2, in X
NameError: name 'blah' isn't defined
```
now I define a dummy function

```
>>> def dummy(x):
...     return x
...     
...     
... 
>>> dummy(3)
3
```

for some reason, the dummy function isn't seen by list (or dictonary) comprehension, although it is present in globals:

```
>>> [dummy(x) for x in [1,2,3]]
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 1, in <listcomp>
NameError: name 'dummy' isn't defined
>>> dir()
['__module__', 'dummy', '__qualname__']
>>> 
```

If I define the function before producing the exception on the class level, then this problem doesn't occur. It looks like after having met a class-level exception, the new function definitions don't stick properly :)

I tried this on ESP32 and Pico (both with MP 1.15) and the behavior was the same.

dpgeorge, 2021-04-27:
Thanks for the report, I can reproduce it on the unix port.  I think it's because the globals dict is not restored if there's an exception in the class body.

It's pretty rare to have exceptions in a class body, so not sure this needs to be fixed.  What's the use case here?

aivarannamaa, 2021-04-28:
> It's pretty rare to have exceptions in a class body, so not sure this needs to be fixed. What's the use case here?

Thonny users occasionally do make mistakes at the class level. After each run Thonny updates its globals table and executes a dict comprehension for this, which uses a hidden helper function set up by Thonny. Before each run Thonny makes a soft reboot, so it needs to restore the helper function after the run completes, and because of the bug the helper remains invisible for the comprehension.

aallan, 2021-05-17:
This is something users run into, eg see raspberrypi/pico-micropython-examples#35, which isn't the first time I've seen this reported.

dpgeorge, 2021-05-18:
It would be nice to fix it but it's expensive in RAM (stack) and CPU usage because of the exception buffer (nlr_buf_t) that must be pushed/popped while running the class body code... not sure if there's a way to make it more efficient though.

lurch, 2021-05-18:
Does the expensive CPU-usage matter if it only happens during an Exception ? (which should be a pretty rare and/or fatal event anyway?) :shrug: Or would the push/pop need to happen _every_ time you run class body code?

dpgeorge, 2021-05-20:
> Or would the push/pop need to happen _every_ time you run class body code?

That's correct, it needs to be pushed/popped regardless of an exception occurring.  Although the class body code is (usually) only run once on import, typical Python code does have a lot of classes.

aivarannamaa, 2021-05-20:
Do I understand correctly, that only program's start-up time would be affected? In this case a push and pop per class looks like a fair price to pay for less confusion.

stinos, 2021-05-20:
Any loading of code/classes is affected. That is not necessarily at startup only: it's not uncommon to delay imports until they are needed. It's also not just a matter of time, also stack usage.