micropython/micropython #7172: Building for ESP-S2

teuler, 2021-04-27:
Hi
I saw that the ESP32 port now allows for the board option GENERIC_S2. This would be great as I am using a FeatherS2 in a project and would really like to stick with MicroPython instead of changing to CircuitPython ...
So I tried to build it but got the following error:
```
/home/tom/micropython/ports/esp32/usb.c:34:10: fatal error: tusb_cdc_acm.h: No such file or directory
 #include "tusb_cdc_acm.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
```
I am using Ubuntu 20.04 running in the Linux Subsystem of 64bit Windows 10, esp-idf 4.2, and the latest MicroPython code.
Building TINYPICO and GENERIC works out of the box.

Any suggestions are appreciated.
Thanks

robert-hh, 2021-04-27:
You have to use esp-idf release/v4.3 at least, and update the submodules in esp-idf after changing the branch.

teuler, 2021-04-27:
Thanks! I'll try that.

dpgeorge, 2021-04-27:
Yes, you need either IDF `v4.3-beta3` or the latest `master`.

closed 2021-04-30 by dpgeorge