micropython/micropython #7174: freezing uasyncio results in import error on first import

mindforger, 2021-04-27:
i have thrown together a current build of this git for an esp8266
into the modules folder i threw uasyncio with __init__, core, quueues, synchro, udp and the websocket folder

when i launch the webrepl eveything is fine and i can do an import uasyncio without any issues, but when i run for the main.py i get this:

Traceback (most recent call last):
  File "main.py", line 98, in <module>
  File "picoweb/__init__.py", line 322, in run
  File "picoweb/__init__.py", line 304, in serve
  File "uasyncio/core.py", line 129, in run_forever
  File "uasyncio/__init__.py", line 33, in add_reader
TypeError: function expected at most 3 arguments, got 4

from this line starting my frozen picoweb server

app.run(debug=False, host=host, port=port)

the calls i find in the uasync modules don't match at all when compared

i have no idea what makes it break but when i leave out the core.py it fails on EVERY first "import uasyncio" with NameError EventLoop not defined any second import will succeed

mindforger, 2021-04-27:
i am stumped ... i did not need to freeze uaynscio at all as it looks like ?! 

feel free to close this but ia was following the picoweb instructions that stated i need that module

and using it as a file resulted in heap fragmentation shunting every 4 or 5 get requests that used a render_template

i am still a little bit confused on what is happening here, if one could maybe shed a bit light on my adventure?

Krasn4ck, 2021-04-27:
> i have thrown together a current build of this git for an esp8266
> into the modules folder i threw uasyncio with **init**, core, quueues, synchro, udp and the websocket folder
> 
> when i launch the webrepl eveything is fine and i can do an import uasyncio without any issues, but when i run for the main.py i get this:
> 
> Traceback (most recent call last):
> File "main.py", line 98, in
> File "picoweb/**init**.py", line 322, in run
> File "picoweb/**init**.py", line 304, in serve
> File "uasyncio/core.py", line 129, in run_forever
> File "uasyncio/**init**.py", line 33, in add_reader
> TypeError: function expected at most 3 arguments, got 4
> 
> from this line starting my frozen picoweb server
> 
> app.run(debug=False, host=host, port=port)

@mindforger  may be missing more arguments in your function run():
`def run(self, host="127.0.0.1", port=8081, debug=False, lazy_init=False, log=None):`

mindforger, 2021-04-27:
those are optionals and not the source of the issue :)
it's more a confusing difference between EventLoop from the core.py and the replacing PollEventLoop that uses additional sockets and the fact that the uascnio init seems to deliberately replace the original core event loop class

mindforger, 2021-04-28:
well nevermind, seems like the issues are already known
https://github.com/pfalcon/picoweb/issues/45

closed 2021-04-28 by mindforger