micropython/micropython #7179: Missing country settings for ESP WIFI

karfas, 2021-04-28:
In micropython, there is no way to define in which country the ESP WIFI transmitters operate.
This might create legal problems for anyone using the ESP devices and micropython "out of the box".

The ESP-IDF function call is esp_wifi_set_country(), described in https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#wi-fi-country-code.

Current uPy sources have no call to esp_wifi_set_country() - I searched *.[hc] for it.

The issue had been discussed in https://forum.micropython.org/viewtopic.php?f=18&t=10388

andypiper, 2022-05-02:
Interestingly, per #8417 the country option is available in wifi on the pyboard / stm port, but apparently not exposed in the ESP port as of now.

jonnor, 2024-09-13:
This still seems to be relevant. Anyone wants to make a MR to add it?