micropython/micropython #7185: [ESP32]urequests, umqtt  not existed in esp32-20210418-v1.15.bin

codemee, 2021-04-29:
The urequests module is not found in esp32-20210418-v1.15.bin firmware. 

[image]

mcauser, 2021-04-29:
Does this work for you?
```
import upip
upip.install('urequests')
```

codemee, 2021-04-29:
It works. But the urequests and umqtt modules are builtin in V1.14 and previous version.

mcauser, 2021-04-30:
Having a look at the 2x manifests:
https://github.com/micropython/micropython/blob/master/ports/esp32/boards/manifest.py
https://github.com/micropython/micropython/blob/master/ports/esp32/boards/manifest_release.py

`urequests` and `umqtt` are present in `manifest_release.py` but not `manifest.py`.

Are you getting your `esp32-20210418-v1.15.bin` from https://micropython.org/download/esp32/ ?

codemee, 2021-04-30:
Yes, I got the file from the download page.

snympi, 2021-06-15:
I have found the same issue moving to V1.15 from V1.14 with `umqtt` that is missing. It is missing from the official download used to flash my esp32.

dpgeorge, 2021-06-17:
This is now fixed, the build system properly selects `manifest_release.py` (it was due to the change to cmake).

closed 2021-06-17 by dpgeorge