micropython/micropython #7188: rp2 - irq_disable irq_enable - sleep for more than 10 micro seconds - rp2 locks

kevindawson, 2021-04-29:
**rp2** Hacking around with irq_disable/irq_enable
inspired by [Writing interrupt handlers](https://docs.micropython.org/en/latest/reference/isr_rules.html#critical-sections)
> This is an instance of a hazard known as a race condition: the ISR and the main program loop race to alter the variables. To avoid inconsistency a means must be employed to ensure that the ISR does not alter the values for the duration of the critical section. One way to achieve this is to issue pyb.disable_irq() before the start of the section, and pyb.enable_irq() at the end.

compleat code sample:
```python
# Passive Infrared Sensor (pis)

import micropython
micropython.alloc_emergency_exception_buf(100)

from machine import Pin, lightsleep, disable_irq, enable_irq, soft_reset
from utime import sleep, sleep_ms, sleep_us

escape_count = 0
led = Pin(25, Pin.OUT)
# Then set up the pin you wired your HC-SR501 sensor to:
ADC0 = 26
pir = Pin(ADC0, Pin.IN, Pin.PULL_UP)

def action_plus_ips_resetting(placebo):
    print("going to sleep", end=" ")
    global escape_count
    escape_count += 1
    for _ in range(8):
        print("z", end="")
        led.toggle()
        sleep(1)
    print("z!")

def pir_handler(pin):
    # prevent the alarm from being triggered by any jitter in the signal from the PIR sensor.
    # The process of smoothing out fluctuations, as we’ve done here, is known as debouncing.
    sleep_ms(80)

    if pin.value():
        print("\tALARM! Motion detected!")
        micropython.schedule(action_plus_ips_resetting, True)

print("PIS acclimatising")
lightsleep(31000)
print("PIS ready")

pir.irq(handler=pir_handler, trigger=Pin.IRQ_RISING)

while escape_count <= 33:
    print("pondering")
    sleep(1)

    irq_state = disable_irq()
    print("start critical")
    sleep_us(10) # max value 
    escape_count += 1
    print(escape_count)
    print("end critical")
    enable_irq(irq_state)

soft_reset()
```
#### Output as expected
```text
pondering
start critical
25
end critical
pondering
        ALARM! Motion detected!
start critical
26
end critical
going to sleep zzzzzzzzz!
pondering
start critical
28
end critical
pondering
```
### But if you sleep_us() > 10 `rp2` locks
```python
    irq_state = disable_irq()
    print("start critical")
    sleep_us(10) # max value 
    escape_count += 1
    print(escape_count)
    print("end critical")
    enable_irq(irq_state)
```
sleep for more than 10 micro seconds and **rp2** locks up.

is this me?
or an issue

robert-hh, 2021-04-30:
> is this me?
> or an issue

I did not dig fully through the code, but it looks as it follows the usual path: for short delays, like 10µs, a busy wait is used. For longer delays a non-busy wait is set-up, using an alarm, which requires interrupts to be enabled. If the interrupt is disabled, this wait never terminates.

kevindawson, 2021-04-30:
@robert-hh 
Thanks for the enlightenment
> for short delays, like 10µs, a busy wait is used. For longer delays a non-busy wait is set-up, using an alarm, which requires interrupts to be enabled. If the interrupt is disabled, this wait never terminates.

will the longer delays threshold move with over/under clocking?

robert-hh, 2021-04-30:
If in understand it right, the actual argument value is used for the decision. That will then have an effect on the wait time for busy waiting.