micropython/micropython #7204: Hang when accessing filesystem if using timers/interrupts

rkarlsba, 2021-05-01:
Hi all

I have an rpi pico loaded with micropython and I have this issue detailed on the links below

https://www.raspberrypi.org/forums/viewtopic.php?f=146&t=310648
https://forum.micropython.org/viewtopic.php?f=21&t=10430

It is reproducable on current micropython by others and it seems to be a bug. Please help :)

robert-hh, 2021-05-02:
A short script causing that error is below:
```
from machine import Pin, Timer
import uos, time

led = Pin(25, Pin.OUT)
timer = Timer()

count = 0

f = open("afile", "w")
f.write("This is a file\n")
f.close()
try:
    uos.remove("bfile")
except:
    pass
    
def blink(timer):
  global count; count = count + 1
  led.toggle()

timer.init(freq=4, mode=Timer.PERIODIC, callback=blink)

while True:
    uos.rename("afile", "bfile")
    time.sleep(0.1111)
    uos.rename("bfile", "afile")
    time.sleep(0.1111)
    print("count =", count)
```

peterbeadle, 2021-07-12:
I can confirm it is reproducible with a differet code base. I'm having issues with both pin interrupts and timers causing random hangs when writing to internal flash or external flash card on spi. Coder works fine (PID controller runs for days) without flash writes.

iabdalkader, 2022-02-21:
I think this was fixed with #7683