micropython/micropython #7206: ubluetooth gattc_read only returning 64 bytes

sheepymeh, 2021-05-03:
I am using an ESP32. I am able to connect and read the characteristic value using gattc_read, but the output is always truncated to 64 bytes. I have tried setting the MTU to 256 and 512. Before setting the MTU, the limit was 20 bytes, but it is always 64 bytes after setting the MTU to a higher value.

Could someone point me to why this limit seems to be in place? Thank you!

dpgeorge, 2021-05-03:
What BLE device is at the other end of the connection?  Can you post minimal code to show the issue?

sheepymeh, 2021-05-03:
I'm trying to understand the [BlueTrace protocol](https://bluetrace.io/static/bluetrace_whitepaper-938063656596c104632def383eb33b3c.pdf), so the other end is an Android phone with the app installed. The ESP32 is acting as the central and is reading the characterstic value from the phone.

On page 3 of the linked PDF is the expected characteristic value, but I am only able to get the first 64 bytes (``{"id":"xx..."``). Here is the related part of the code I am using:

```python
# UUIDs were taken from the OpenTrace repository
elif event == _IRQ_PERIPHERAL_CONNECT:
	tt_devices[data[0]] = [data[2]]
	bt.gattc_discover_services(data[0], ubluetooth.UUID('...'))

elif event == _IRQ_GATTC_SERVICE_RESULT:
	tt_devices[data[0]].extend([data[1], data[2]])

elif event == _IRQ_GATTC_SERVICE_DONE:
	if len(tt_devices[data[0]]) == 3:
		# Should follow this path
		print('TraceTogether device found')
		bt.gattc_discover_characteristics(data[0], tt_devices[data[0]][1], tt_devices[data[0]][2], ubluetooth.UUID('...'))
	else:
		print('Discarding non-TT device')
		bt.gap_disconnect(data[0])
		del tt_devices[data[0]]

elif event == _IRQ_GATTC_CHARACTERISTIC_RESULT:
	tt_devices[data[0]].append(data[2])

elif event == _IRQ_GATTC_CHARACTERISTIC_DONE:
	if len(tt_devices[data[0]]) == 4:
		# Should follow this path
		print('TraceTogether characteristic found')
		bt.gattc_exchange_mtu(data[0])
	else:
		print('Discarding non-TT characteristic')
		del tt_devices[data[0]]
		bt.gap_disconnect(data[0])

elif event == _IRQ_MTU_EXCHANGED:
	print('MTU exchanged')
	# This sets the MTU to 512 and reads the characteristic value
	bt.gattc_read(data[0], tt_devices[data[0]][3])

elif event == _IRQ_GATTC_READ_RESULT:
	# This line is only able to read the first 64 bytes. len(data[2]) returns bytes as well
	print(bytes(data[2]).decode('utf-8'))
```

sheepymeh, 2021-05-06:
Seems like this was because I had forgotten to set ``bt.config('rxbuf')`` to a larger value.

closed 2021-05-06 by sheepymeh