micropython/micropython #7207: Documentation WLAN.connect()

karfas, 2021-05-03:
In https://docs.micropython.org/en/latest/wipy/tutorial/wlan.html, the connect() method reads 
`wlan.connect(net.ssid, auth=(net.sec, 'mywifikey'), timeout=5000)` 

In contrary, https://docs.micropython.org/en/latest/library/network.WLAN.html?highlight=wlan#network.WLAN.connect describes the connect() call as
`WLAN.connect(ssid=None, password=None, *, bssid=None)`

Which one is the correct/current one (and for which port ?) ?
This is quite confusing for new users, I think.

karfas, 2021-05-03:
Additional, the `wlan.mode()` call from the tutorial does not exist in library reference.

WaGi-Coding, 2022-07-13:
bssid seems to not even be a valid argument

andypiper, 2022-07-13:
@WaGi-Coding what kind of board are you using / which MicroPython port are you using?

Some of these function calls or parameters may be port-specific.

WaGi-Coding, 2022-07-13:
@andypiper I just figured out I was was passing the parameters wrongly, bssid is the only one you can/have To pass with the parameter name, I tried all without or all with. It's all fine now

jimmo, 2022-07-14:
Yeah the docs are wrong

`.. method:: WLAN.connect(ssid=None, key=None, *, bssid=None)`

should read

`.. method:: WLAN.connect(ssid=None, key=None, /, bssid=None)`

(Note the * changes to /, as per https://peps.python.org/pep-0570/#how-to-teach-this )

Working on docs at the moment. I had already fixed the discrepancy between the tutorials and the module docs, have also added this change too.