micropython/micropython #7208: Pyboard-D: Webrepl not responding

kevinkk525, 2021-05-03:
I tried using the webrepl on my pyboard-D but it doesn't go further than entering my password. After that, nothing happens. Not even the repl output gets sent to the webrepl. Sending/Receiving files also doesn't work.
When I interrupt my uasyncio program or any other executed code on the pyboard, everything works fine again. (But I need to Interrupt from the console, it doesn't work through the webrepl)
Oddly enough, starting the same code again from the webrepl makes the repl output appear in the webrepl, however the KeyboardInterrupt still won't work.
I don't have these problems on my esp8266 or esp32.

I'm using micropython 1.15 nightly from the download page.

additionally: Sending files to the pyboard through webrepl doesn't make them appear in the PYBFLASH drive on windows until a hard reset. Not even a soft reset makes them show up. But that is a separate issue and not really important (who uses webrepl to upload files while connected to a pc...)

Steps to reproduce:

1. Connect to wlan and start webrepl
2. Have a script (or run from serial repl) start some code:
```
import time
c=0
while True:
    print(c)
    c+=1
    time.sleep(1)
```
3. Connect through webrepl
4. Entering password works, serial repl prints connection established

Expected Behaviour:
The output of the script gets printed to both serial repl and webrepl, file upload/downloads work while the script runs, KeyboardInterrupts work.

Actual Behaviour:
1. No output printed to the webrepl. Interestingly though, if you Interrupt the script from the serial repl and start it again from the webrepl, the output gets printed to the webrepl. But reconnecting the webrepl results in the same problem, no output.
2. No KeyboardInterrupt. Not even when the script prints the output to the webrepl.
3. No file upload/download while a script is running.
4. Disconnect only due to timeout while a script is running.

robert-hh, 2021-05-03:
> additionally: Sending files to the pyboard through webrepl doesn't make them appear in the PYBFLASH drive on windows until a hard reset. 

That's a PC behvior and a common cause for file system corruption on the board. The PC's OS does not expect the board to change the content of it's file system. The PC assumes it's the only instance doing so. A reset detaches the USB shortly and causes the PC to re-scan the board.

kevinkk525, 2021-05-11:
Judging by the lack of responses here and in the forum thread, I must be the only one with this problem. So for everyone else reading this, your webrepl is working just fine on the pyboard?

felixdoerre, 2022-09-13:
Just to post for future reference, it seems that the feature to have KeyboardInterrupts accepted while not reading from stdin (like while waiting with sleep or reading from a socket), is only implemented for esp8266:
 https://github.com/micropython/micropython/blob/4903e48e340bd9741577f30cacb31605cc70cf20/ports/esp8266/moduos.c#L64
in combination with https://github.com/micropython/micropython-lib/blob/796a5986cd65e8f2c76d06267b92497ff602eca6/micropython/net/webrepl/webrepl.py#L132. Although other ports could probably quite easily import that callback as well.

felixdoerre, 2022-09-13:
Ooh, I just found 2 more ports implementing this callback. And it seems they do it in a rather "portable" way:
https://github.com/micropython/micropython/blob/bd4e45fd68e20af3f965b403e643fa5f71aa1b08/ports/esp32/moduos.c#L53

Probably the respective function could just be copied to the port you're interested in.