micropython/micropython #7213: Pyboard D doesn't seem to do anything after building and flashing

Corjan85, 2021-05-04:
Hi,

I'm probably missing a vital step here, but after both building the micropython st32 port to the pyboard D I don't get any response from it. It seems that it should at least create a virtual COM port right for the REPL interface?

Things I did:
git checkout v1.15
make BOARD=PYBV11
make BOARD=PYBV11 USE_PYDFU=0 deploy

Both building and flashing seem to go fine. If I don't specify 'USE_PYDFU=0', it doesn't seem to want to flash.

Please let me know if more information is required,

Corjan

robert-hh, 2021-05-04:
Which pyboartd is your target? you have built the firmware for pyboard V1.1, but for  Pyboard D you have to specify either BOARD=PYBD_SF2, BOARD=PYBD_SF3 or BOARD=PYBD_SF6, depending on the model.

Corjan85, 2021-05-04:
Hi,

Thats it!

Thanks for your quick response, and sorry for wasting your time :).

Corjan

Corjan85, 2021-05-04:
While you might be

closed 2021-05-04 by Corjan85