micropython/micropython #7218: SPI clock like with CPOL=1 (polarity=1) does not initially idle high (rp2)

ShawnHymel, 2021-05-04:
I'm working on a Raspberry Pi Pico (RP2040) and trying to communicate with an Adafruit ADXL343 breakout board. The SPI setup requires CPOL=1 and CPHA=1, which means the clock (SCK) line should idle high. When looking at the communication on a logic analyzer, I see that SCK idles low initially until the first byte has been sent. After that, it idles high. This means the first byte sent is corrupted and often not interpreted correctly by the receiver (as in my case).

I either have to throw away the first byte or manually set the SCK pin high before transmission. Code and screenshot of logic analyzer below. Please note that I have not tested this on other MicroPython ports.

```
import machine
import utime

# Assign chip select (CS) pin (and start it high)
cs = machine.Pin(17, machine.Pin.OUT)
cs.value(1)

# Initialize SPI peripheral
spi = machine.SPI(0,
                  baudrate=500000,
                  polarity=1,
                  phase=1,
                  bits=8,
                  firstbit=machine.SPI.MSB,
                  sck=machine.Pin(18),
                  mosi=machine.Pin(19),
                  miso=machine.Pin(16))

# Workaround: need to set SCK high manually

# TEST: read from ID register
msg = bytearray()
msg.append(0x80) 

# Send out SPI message and read
cs.value(0)
spi.write(msg)
data = spi.read(1)
cs.value(1)
print(data)

# TEST: read from ID register
msg = bytearray()
msg.append(0x80) 

# Send out SPI message and read
cs.value(0)
spi.write(msg)
data = spi.read(1)
cs.value(1)
print(data)
```

[image]

chrismas9, 2021-05-04:
It's the same on the STM32 port. I always do a single byte write after setting the mode before asserting the first nCS. I have multiple SPI devices with different polarities so I have to do it every time I access a different peripheral.

kevindawson, 2021-05-04:
@ShawnHymel 

have you tried Pin.PULL_UP

```python
# Assign chip select (CS) pin (and start it high)
cs = machine.Pin(17, machine.Pin.OUT, machine.Pin.PULL_UP)
```

ShawnHymel, 2021-05-04:
@kevindawson That's a good idea, but it doesn't work in this case. I need SCK to idle high (not CS). I tried adding it to my SPI init:

```
spi = machine.SPI(0,
                  baudrate=500000,
                  polarity=1,
                  phase=1,
                  bits=8,
                  firstbit=machine.SPI.MSB,
                  sck=machine.Pin(18, machine.Pin.OUT, machine.Pin.PULL_UP),
                  mosi=machine.Pin(19),
                  miso=machine.Pin(16))
```

However, that does not seem to correct the issue. SCK still starts low until the first byte has been clocked.