micropython/micropython #7220: BLE: Advertising name stays as 'ESP32'

plugowski, 2021-05-05:
I have an issue while advertise my device, I used ble_advertise helper from examples, which  generates payload as below: 

`payload = b'\x02\x01\x06\x05\tLEDs\x11\x07\xdb\xa7\x7fgl\xbfG\x9c\x89@\x16\x05S@\x8a<'`

So visible name in Bluetooth scanners should be **LEDs** but it is advertising as ESP32 all the time (I saw LEDs, few times only, after that it has been override as ESP32).

Any suggestions?

dpgeorge, 2021-05-05:
You might need to use:
```python
ble.config(gap_name="myname")
```
Also, some scanners will save names of previously seen devices, so you'll need to find a way to reset them.

plugowski, 2021-05-05:
Okay I'll try that, I saw it in docs, but thought that advertise will override it, so it's not necessary to setup it in two places...

dpgeorge, 2021-05-05:
The GAP name is independent to the advertising string, and there are probably cases where it makes sense for them to be different.

plugowski, 2021-05-10:
I renamed device in two places using config and address, and it's still like random, sometimes ESP32, and sometimes name which I setup. Tested on different BLE scanners for mac and iOS/

agatti, 2026-05-23:
This was fixed in d014c8282652d0e2643a9ba088445c542ad43450, closing as completed.

closed 2026-05-23 by agatti