micropython/micropython #7225: WebREPL doesn't seem to work on SAOLA

mishafarms, 2021-05-06:
I created a board directory for SAOLA (without the USB) compiled. It seems to run fine from the serial port REPL, but when I enabled the WebREPL, I do not get very far. Sometimes it just disconnects right away sometimes it lets me type a few characters.
I am using esp-idf commit c8315e01107c10898947a817d5cf763cdd3f5a4f (HEAD, tag: v4.4-dev)

Here is the script I am using to test.

`import network
import utime
import webrepl

def start_net():
    sta = network.WLAN(network.STA_IF)
    sta.active(True)
    sta.connect("Mishafarms", "deadbeef01")
    while not sta.isconnected():
        utime.sleep(1)

start_net()
webrepl.start(password="")
`

Here is the output I get from the serial REPL.

WebREPL daemon started on ws://192.168.252.210:8266
Started webrepl in manual override mode
>>> 
WebREPL connection from: ('192.168.252.162', 34592)
dupterm: Exception in read() method, deactivating: OSError: [Errno 113] ECONNABORTED

WebREPL connection from: ('192.168.252.162', 34594)

>>> 
>>> 
>>> dupterm: Exception in read() method, deactivating: OSError: [Errno 113] ECONNABORTED

WebREPL connection from: ('192.168.252.162', 34596)
prdupterm: Exception in read() method, deactivating: OSError: [Errno 113] ECONNABORTED

robert-hh, 2021-05-06:
I had that in the ESP32S2 port. The reason was too little memory for the WiFi stack. Since you have built teh firmware yourself, you could try to reduce the amount of RAM assigned to the Python heap. I reduced it by 16k, but less may also help. The code for that is in main.c. It is a line like:
`size_t mp_task_heap_size = heap_caps_get_largest_free_block(MALLOC_CAP_8BIT);`
change it to
`size_t mp_task_heap_size = heap_caps_get_largest_free_block(MALLOC_CAP_8BIT) - 16 * 1024;`

mishafarms, 2021-05-06:
Thanks for the feedback, I will try that right away.

mishafarms, 2021-05-06:
I tried this with the PR#7163 and it seemed to still happen. I will try it with PR#7222 and see.

Thanks again

robert-hh, 2021-05-06:
There are a few lines like this. You have to find the right one in the #if section for esp32s2.

mishafarms, 2021-05-06:
I enabled SPIRAM and I get the problem. Since the spiram doesn't use the malloc, I would assume that there would be plenty of
ram for the WiFi stack. I checked to make sure it was using the SPIRAM for the python heap using gc.mem_free().

mishafarms, 2021-07-01:
This is still failing when I try it with SPIRAM enabled. Any chance of someone looking at it. It is beyond my abilities.

Michael

ThinkTransit, 2021-08-21:
I'm getting the exact same issue on the UM_FEATHERS2 board. Did you find a solution Michael?

Thanks

Maddox-zephyr, 2021-08-28:
I am having the same problem on UM_FEATHERS2 board as well.  Does it work on any S2 boards? That we know of?

dpgeorge, 2021-09-04:
See related #7726.

ThinkTransit, 2021-09-05:
> See related #7726.

Thanks works on ESP32-S2-WROVER when MP compiled with tinyusb disabled.

mishafarms, 2021-11-09:
I think I have found the cause for the browser failure to connect to network REPL.
hashlib.sha1() is not giving the correct hash on the S2 device. This causes the browser to not get back the correct 
Sec-WebSocket-Accept value.

The simple way to test this is this code.
```
import hashlib

print("".join("\\x%02x" % i for i in hashlib.sha1("test").digest()))
```

The output of this test should be 
\xa9\x4a\x8f\xe5\xcc\xb1\x9b\xa6\x1c\x4c\x08\x73\xd3\x91\xe9\x87\x98\x2f\xbb\xd3
which I get on the ESP32 port.
But on the ESP32S2 I get 
\x95\xcc\x48\x93\xdb\x8d\xe5\xa0\xcb\xe4\x5e\x93\x5a\x27\xf4\x4c\xd2\xf5\xd7\xb3
which is wrong according to a sha1 test on the web.

If someone has any info on why these are different, that would help.

Michael

mishafarms, 2021-11-10:
A quick solution to the problem is to remove the SHA from hardware by using this line in the sdkconfig.board file.

CONFIG_MBEDTLS_HARDWARE_SHA=n

mishafarms, 2021-11-21:
I have found the root cause.

The problem exists because in the ESP sdk file /components/mbedtls/port/sha/dma/sha.c,
there is code to write out the cache for the external memory, But it is protected with 
#if (CONFIG_SPIRAM_USE_CAPS_ALLOC || CONFIG_SPIRAM_USE_MALLOC)
This is not going to work for micropython, because micropython uses CONFIG_SPIRAM_USE_MEMMAP.
I have tested with adding CONFIG_SPIRAM_USE_MEMMAP, and changing the whole line to CONFIG_SPIRAM.
both work.
I have created an issue on the esp SDK github.

Hansi137, 2022-02-19:
Hey guys i actualy have the same issue when connecting to the webrepl console via wifi over my Android smartphone.

Is there a practical solution for me? I am Just a beginner with esp32 and micropython...

Greetings from germany

ThinkTransit, 2022-02-19:
The command line version of webrepl should work, try this

https://github.com/micropython/micropython/issues/7726#issuecomment-911754886

mishafarms, 2022-02-26:
This is the quick solution
https://github.com/micropython/micropython/issues/7225#issuecomment-965734500