micropython/micropython #7226: "BINARYEN_TRAP_MODE is not supported by the LLVM wasm backend" with javascript port

iizukak, 2021-05-06:
When I tried to build javascript port, I got below error message.

Maybe relate to [this line](https://github.com/micropython/micropython/blob/master/ports/javascript/Makefile#L54) in the Makefile.

## Command & Error Message

```
$ cd ports/javascript && make
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
LINK build/firmware.js
emcc: error: BINARYEN_TRAP_MODE is not supported by the LLVM wasm backend
Makefile:59: recipe for target 'build/micropython.js' failed
make: *** [build/micropython.js] Error 1
```

## Test Environment

### Commit ID

9eea51b73

### clang Version

```
$ clang --version
clang version 6.0.0-1ubuntu2 (tags/RELEASE_600/final)
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
```

### emcc Version

```
$ emcc --version
emcc (Emscripten gcc/clang-like replacement) 1.40.1 (77d7739b50a765c8208c5e6105ef58b6e21ed163)
Copyright (C) 2014 the Emscripten authors (see AUTHORS.txt)
```

### OS Version

```
$ cat /etc/os-release
NAME="Ubuntu"
VERSION="18.04.5 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.5 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic
```

embeddedt, 2021-05-24:
Since you're using WASM you should be able to just remove `-s "BINARYEN_TRAP_MODE='clamp'"` from `JSFLAGS`. This is the fix I used downstream in [lv_micropython](https://github.com/lvgl/lv_micropython/tree/lvgl_javascript/ports/javascript).

The JavaScript port sadly doesn't receive much love as it's not really a good target for Python at the moment. The port is also somewhat incomplete.

dpgeorge, 2021-07-22:
This was fixed by c13853f4dab9c65ee6d1af99b25f999b16ef2827

closed 2021-07-22 by dpgeorge