micropython/micropython #7227: array comparision with different types is implementad as byte comparison

stinos, 2021-05-06:
```
import array

array.array("b", [1, 2]) == array.array("i", [1, 2])
```

MicroPython: `False`
CPython: `True`

Would be better if MicroPython would raise an exception here, instead of silently doing the wrong thing. I'm looking into a fix but it's not trivial because just requiring e.g. typecodes to match is not an option because comparing bytes/bytearray/array('b') should keep working,

stinos, 2021-05-06:
Related:
```
print("123" == bytearray(b"123"))
```
MicroPython: `True`
CPython: `False`

Doesn't seem like a major problem, but should best be documented.

So the question is: should all possible byte comparisons remain as-is?

stinos, 2021-05-10:
Also:
```
print(array.array('d', [float('nan')]) == array.array('d', [float('nan')]))
```
MicroPython: `True`
CPython: `False`

Peculiar, but might require too much code size to fix

dpgeorge, 2021-05-13:
Fixed by 57365d855734142deb030ebcd00c10efcedf554b (raises `NotImplementedError` for types that cannot be easily compared).

closed 2021-05-13 by dpgeorge