micropython/micropython #7228: Compression support

br0kenpixel, 2021-05-06:
Hi. I'm looking for a way to compress long strings in MicroPython.
I found these two pull requests: #5613 and #6972.
Apparently, in these pull requests, GZIP compression support was added, however, when I compiled latest MicroPython source, I could not find a uzlib.gzip() function. Maybe it was never really implemented (or it was removed for some reason).

Is there any alternative to this? Or could I somehow "enable" that function?

br0kenpixel, 2021-05-06:
> What platform are you using? On RP2 `import uzlib` works.

I'm using an ESP32.

jonnor, 2024-07-03:
Recent versions support the `deflate` module, and it is included by default on ESP32. For example on MicroPython 1.23:

```python
import deflate
>>> dir(deflate)
['__class__', '__name__', 'AUTO', 'DeflateIO', 'GZIP', 'RAW', 'ZLIB', '__dict__']
```
So I believe this issue can be closed now :)

closed 2024-07-04 by dpgeorge