micropython/micropython #7229: Module import fails if a directory exists with same name

peterhinch, 2021-05-07:
Tested under Unix and RP2.

To replicate create a directory (e.g. ~/test) containing `foo.py`:
```python
def bar():
    print('bar')
```
Import behaves as expected. CPython behaves identically.:
```
MicroPython v1.14 on 2021-03-28; Raspberry Pi Pico with RP2040
Type "help()" for more information.
>>> import foo
>>> dir(foo)
['__class__', '__name__', '__file__', 'bar']
>>> 
```
Create a subdirectory `foo` and repeat:
```python
MicroPython v1.14 on 2021-03-28; Raspberry Pi Pico with RP2040
Type "help()" for more information.
>>> import foo
>>> dir(foo)
['__class__', '__name__', '__path__']
```
CPython correctly imports the module:
```python
[adminpete@capybara]: ~/test
$ python3
Python 3.8.5 (default, Jan 27 2021, 15:41:15) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import foo
>>> dir(foo)
['__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__spec__', 'bar']
>>> 
```

talbor49, 2021-05-07:
Also interesting - notice how if we add a `__init__.py` file to `foo/`, both will act the same and import the directory(package).
It's almost like CPython has some heuristic that will make it skip an import option if it has no members.

Although I don't think this is a problem, and not a "failure" of module importing. These are just different ways to handle import ordering.

peterhinch, 2021-05-09:
Another manifestation is this:
```python
$ upython 
MicroPython v1.14-128-gca3d51f12-dirty on 2021-03-31; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> from foo import bar
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'foo.bar'
>>> 
```
Again this works OK in CPython. It seems that MP assumes that the directory `foo` is a Python package, looks for `bar` in the package and fails.