micropython/micropython #7233: Unable to compile natmod uzlib example

br0kenpixel, 2021-05-09:
When I try to compile the uzlib example I get the following error:
```
GEN build/uzlib_xtensawin.config.h
CC uzlib.c
LINK build/uzlib.o
LinkError: build/uzlib.o: undefined symbol: realloc
make: *** [build/uzlib_xtensawin.native.mpy] Error 1
```
I'm on MacOS 10.15.7 and using ESP-IDF v4.2

dpgeorge, 2021-05-09:
There shouldn't be any calls to `realloc`, they should all go through `m_realloc_dyn`.

Can you provide a dump of `uzlib.o` via `xtensa-esp32-elf-objdump -xd build/uzlib.o`?

br0kenpixel, 2021-05-09:
Sure thing. I attached the dump.
[dump.txt](https://github.com/micropython/micropython/files/6448075/dump.txt)

dpgeorge, 2021-05-10:
It looks like there is code included in your build for compression, which is not in this repo.  Are you building uzlib from a clean version of this repo?

br0kenpixel, 2021-05-10:
No. I actually modified it. I found another repo with a modified uzlib that supports compression. I didn't think it would affect building the natmod. I wanted to create .mpy of the modified uzlib module so if anyone needs compression support, then they could just download the .mpy file and import it into their code without having to download the source, apply the patches and build it.

jonnor, 2024-07-03:
If the modified code acutally uses realloc, then you can provide a stub defintion of that file, which calls m_realloc_dyn (or similar) from the MicroPython library table.