micropython/micropython #7235: mimxrt: Board crashes on print()

robert-hh, 2021-05-09:
The following statements on REPL:

b = bytearray(1024)
print(b)

cause the board to fall into bootloader mode. The statement `print(b)` or just 'b' must eventually repeated a few times. Sometimes it's pretty stable. The size of the bytearray does not matter, as long as it exceeds about 256 bytes. The larger the print the worse.

dpgeorge, 2021-05-09:
At v1.15, all tests in `tests/basics/` passed on Teensy 4.0.  Can you verify if they pass for you?  (Be sure that git submodules are up to date.)

robert-hh, 2021-05-09:
I used the minimal target. It passes all basics tests, which it does not skip. It freezes on extmod/ticks_diff.py. I de-installed and reinstall again the nxp_driver lib, getting the same version again.

dpgeorge, 2021-05-10:
Ok, I can reproduce this issue here.

The following seems to fix it, but I don't know why:
```diff
--- a/ports/mimxrt/mphalport.c
+++ b/ports/mimxrt/mphalport.c
@@ -87,7 +87,7 @@ void mp_hal_stdout_tx_strn(const char *str, mp_uint_t len) {
                 n = CFG_TUD_CDC_EP_BUFSIZE;
             }
             while (n > tud_cdc_write_available()) {
-                __WFE();
+                tud_cdc_write_flush();
             }
             uint32_t n2 = tud_cdc_write(str + i, n);
             tud_cdc_write_flush();
```

robert-hh, 2021-05-10:
Here it's not fixed. Sometimes it works quite a while, and then after a power cycle it breaks immediately. So it looks more like cause by a non-initiaized value, or a timing problem. Try a larger bytearray size, like 4096.
So at least it is consistent: If YOU do not know why something fixes a problem, it's most likely not a fix.

robert-hh, 2021-05-10:
Since that also happens when just creating the bytearray, it looks more like a problem with dynamic memory management.

dpgeorge, 2021-05-18:
I've been trying to get to the bottom of this but had no luck yet.  Indeed it seems related to WFI/WFE but not sure how that is affecting the USB.

dpgeorge, 2021-06-03:
I think related to this is commit 5226d6e1ee7282bf40a45708b98067446b157bce which removed `__WFE()` from `MICROPY_EVENT_POLL_HOOK`.  @robert-hh can you provide instructions how to reproduce the error you see when WFE is included in `MICROPY_EVENT_POLL_HOOK`?

robert-hh, 2021-06-03:
It used to crash when printing large amounts of data, or sometimes smaller amount. But I cannot reproduce it with the current master. I can reproduce it with the UART module, being about to published at https://github.com/robert-hh/micropython/tree/mimxrt_uart, which is howver not yet rebased on the current master. In the wait loop at read() and write() it calls MICROPY_EVENT_POLL_HOOK. A simple script:
```
from machine import UART
uart=UART(1)
uart.write("hello")
uart.read()
```
stalls, if __WFE() is included, and runs fine without it. It works fine too, if I for instance use uart.read(uart.any()). The analysis in the debugger showed, that using read() it times out once returning 5 symbols read, and when then called again, it seems to stall in the MICROPY_EVENT_HOOK. That is hard to tell since the debugger creates events, but slows down the timer.

dpgeorge, 2021-06-03:
All IRQ handlers that call `mp_sched_schedul()` should also call `__SEV();` to indicate there is a new "event" to process and to wake up the `MICROPY_EVENT_POLL_HOOK`.

robert-hh, 2021-06-03:
I understand, but at the time when the code stalls there is no event happening in this context, besides maybe the timer tick, because no more symbols arrive at the UART.

robert-hh, 2021-06-03:
If I add the __SEV(); call to the uart polling loop, creating a fake event, it works with __WFE(). You did that already in the timer tick ISR.
 Code snippet below:
```    for (size_t received = 0; received < size; ) {
        // Wait for the first/next character
        while ((avail = LPUART_TransferGetRxRingBufferLength(self->lpuart, &self->handle)) <= 0) {
            if (ticks_us64() > t) {  // timed out
                if (received <= 0) {
                    *errcode = MP_EAGAIN;
                    return MP_STREAM_ERROR;
                } else {
                    return received;
                }
            }
            __SEV();
            MICROPY_EVENT_POLL_HOOK
        }
        // get as many bytes as possible to meet the need
        nget = avail < (size - received) ? avail : size - received;
        xfer.data = dest + received;
        xfer.dataSize = nget;
        LPUART_TransferReceiveNonBlocking(self->lpuart, &self->handle, &xfer, NULL);
        received += nget;
        t = ticks_us64() + timeout_char_us;
    }
```

robert-hh, 2021-06-03:
Still a problem. After I rebased the UART branch on the current master, my Teensy 4.0 would not even boot with __WFE() in the MICROPY_EVENT_POLL_HOOK. It falls back into firmware load mode.

dpgeorge, 2022-01-27:
Is this now fixed/resolved?

robert-hh, 2022-01-27:
Nothing changed since then. Still no __WFE() in the MICROPY_EVENT_POLL_HOOK. 
I simply might have lost it out of focus.  So maybe better keep it for remembering, even if it may be hardly visible if buried deep in the history.

dpgeorge, 2022-01-27:
I would like to eventually fix this, or at least understand more about WFI/WFE on mimxrt.

robert-hh, 2022-01-27:
Coming back to the initial discussion:
> All IRQ handlers that call mp_sched_schedul() should also call __SEV(); to indicate there is a new "event" to process and to wake up the MICROPY_EVENT_POLL_HOOK.

There are two places where mp_sched_schedule() are called from the mimxrt code: machine_time.c and shared/runtime/mpirq.c. None of them calls __SEV(). Nevertheless, adding __WFE() to MICROPY_EVENT_POLL_HOOK still does not work. The code seems not to crash any more at large prints, but it still stalls at uart.read().

robert-hh, 2022-10-22:
After a long break, I searched for that topic again and found a post with a fix. It is required for boards with RT1052, RT1062 and most likely  RT1064,  but not for RT1010, RT1015 and RT1020.
The post: https://community.nxp.com/t5/i-MX-RT/Problem-with-Systick-after-WFI-on-a-RT1062/m-p/1542259#M22176

robert-hh, 2022-10-23:
The issue is covered in PR #9721.

closed 2022-10-23 by robert-hh

dpgeorge, 2022-10-26:
> After a long break, I searched for that topic again and found a post with a fix.

Fantastic!