micropython/micropython #7236: QSTR build phase include path problem with TinyUSB headers

kentindell, 2021-05-09:
RP2 code that uses headers from TinyUSB doesn't get the include path set right when the QSTR phase is run. Something is screwy with the maze of CMake. You can check this by adding:

```
#include "tusb.h"
```

at the top of ``modmachine.h`` in the RP2 port and it barfs during make. I'm not able to work out a proper fix, but I got a bodge workaround by adding ``-I${PICO_TINYUSB_PATH}/src`` to the following line in ``mkrules.cmake`` in the ``py`` directory:

```
    COMMAND ${Python3_EXECUTABLE} ${MICROPY_PY_DIR}/makeqstrdefs.py pp ${CMAKE_C_COMPILER} -I${PICO_TINYUSB_PATH}/src -E output ${MICROPY_GENHDR_DIR}/qstr.i.last cflags ${MICROPY_CPP_FLAGS} -DNO_QSTR cxxflags ${MICROPY_CPP_FLAGS} -DNO_QSTR sources ${MICROPY_SOURCE_QSTR}
```

dpgeorge, 2021-05-10:
Actually, this is a much better fix:
```diff
--- a/ports/rp2/CMakeLists.txt
+++ b/ports/rp2/CMakeLists.txt
@@ -154,6 +154,7 @@ set(PICO_SDK_COMPONENTS
     pico_sync
     pico_time
     pico_unique_id
+    tinyusb_common
     tinyusb_device
 )
 
```

dpgeorge, 2021-05-11:
Fixed by 4404dababb715916262ca9aa45a5732d8454eda0

closed 2021-05-11 by dpgeorge

kentindell, 2022-04-07:
I'm afraid this bug has come back again: if you include ``tusb.h`` at the top of ``modmachine.c`` you'll see it barf in the QSTR phase.