micropython/micropython #7260: doc: state that some function are not built into released firmware.

pidou46, 2021-05-13:
Hello,

It's not mention in the doc that some functions are not included in the firmware, but available to build custom firmware.
It would be good to mention it one way or another to avoid repeatedly asking on the forum.
I think it should be taken into account for GSOC on documentation.
 
see forum topic:

https://forum.micropython.org/viewtopic.php?f=18&t=10503&p=58035#p58035

dlech, 2021-05-13:
I was just thinking yesterday that something like the `Availability` tag in the Python docs would be nice for this ([example](https://docs.python.org/3/library/signal.html)).

In MicroPython, we could use this to state the required compile flag(s) for a feature and perhaps which ports/boards where they are enabled by default.

mattytrentini, 2021-06-03:
Good points! We've already discussed some of those topics internally, here's the summary:

@pidou46 we'd actually like to document _all_ the defines available when building - there are quite a lot available to customise _many_ aspects of the build. Additionally it would then be good to allow them to be configured with something like [menuconfig](https://en.wikipedia.org/wiki/Menuconfig). It might be possible to at least document them as part of GSoD 2021. We have a weekly meeting to discuss progress, I'll bring it up.

@dlech We've been considering something like the `Availability` tag for some time (we were looking into using icons to represent each of the ports). We'd ideally like to create generic API documentation - particularly for `machine` - and then have a way of marking up which ports support the standard methods or how they differ (maybe as a slide-out for notes). It's not a trivial task - and we're constrained by limitations in sphinx -  but it has been discussed for GSoD 2021. 
We hadn't discussed cross-referencing the relevant defines to the API docs but it's a good idea, especially if we can figure out how to do so without being too _noisy_ (many users aren't interested in building so I'd rather not make the defines prominent).

Thanks for the input folks, I'll do my best to get some of this implemented and keep you posted.