micropython/micropython #7261: memoryview(memoryview()[slice]) and gc

jepler, 2021-05-13:
I was investigating https://github.com/v923z/micropython-ulab/issues/387 and discovered that there is a scenario where use of "memoryview of a slice of a memoryview" can lead to misbehavior.

Testing at 43e7e5f00a601b36b5b39f1fef036d6c25d5562c with a unix build and the following script:
```python
import array
import gc
data1 = memoryview(memoryview(array.array('f', [1] * 1000))[6:-6])
print(sum(data1)) # should print 988, consistently does
print(list(data1[:12])) # should print an array full of ones, consistently does
[1,2,3,4,5,6,7,8]
gc.collect()
print(sum(data1)) # should print 988, DOES NOT
print(list(data1[:12])) # should print an array full of ones, DOES NOT
```

```
$ ./micropython  gcproblem2.py 
988.0000000000001
[1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0]
-1.019102854101721e+37
[0.0, 0.0, -9.799065904824237e+34, 3.067302208560592e-41, 0.0, 3.544921875, 0.0, 0.0, 0.0, 0.0, -9.799065904824237e+34, 3.067302208560592e-41]
$ ./micropython  gcproblem2.py 
988.0000000000001
[1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0]
1.774877057396014e+20
[0.0, 0.0, 6.858390406705439e-07, 3.061276625163996e-41, 0.0, 3.544921875, 0.0, 0.0, 0.0, 0.0, 6.858390406705439e-07, 3.061276625163996e-41]
```

jepler, 2021-05-13:
I suspect that the trick where a memoryview must keep the original storage "live" is not working in this case.

dpgeorge, 2021-05-14:
Thanks for the report.  See #7264 for a WIP fix.

dpgeorge, 2021-05-18:
Fixed by 6d2680fa36696fd0b2fd7a595f4a26153cd84e07

closed 2021-05-18 by dpgeorge