micropython/micropython #7269: modffi int handling in 32 bit architecture, in big endian architectures, and with unsigned values

abiliojr, 2021-05-16:
While trying to use modffi, I found the following issues:

- integers of 8 and 16 bits don't work on big endian
- integers of 64 bits don't work on 32 bits architectures
- unsigned returns were converted to signed python integers

A proposed fix for all these issues was submitted as PR  #7268

Please see [this zip](https://github.com/micropython/micropython/files/6488753/bad.int.mwe.zip) for a minimal working example.

dlech, 2021-05-16:
It would be useful to include the tests in the pull request.

abiliojr, 2021-05-16:
No problem adding some tests, just not sure how to test big endian (MIPS, PowerPC). Please give any suggestions you might have.

Also, is there any example of a test that requires compiling some additional C code? It might be really useful

dlech, 2021-05-16:
If we want automated CI tests for big-endian CPUs, that could probably be done by running executables with QEMU (using docker if a full big-endian rootfs is needed).

For the C code, have a look at https://github.com/micropython/micropython/blob/master/ports/unix/coverage.c and https://github.com/micropython/micropython/blob/master/ports/unix/coveragecpp.cpp. We could probably add a similar `coverage-ffi.c` file that contains simple functions for ffi to call.

abiliojr, 2021-05-16:
Though I manually tested the changes using a MIPS, I believe it would be convenient to try the QEMU way, provided it's not too much work. Do you have any experience doing such thing with AppVeyor?

dlech, 2021-05-16:
I've built and run Arm stuff on Travis CI both with and without docker. I'm guessing that we would need a docker image so that we get a big-endian CPython as well for tests.

Back then, I make an Arm docker image from scratch for local development and use the same on CI. But nowadays, docker has pre-made multiarch images. This looks like a good place to start: https://docs.docker.com/desktop/multi-arch/

It is basically a matter of deriving a docker image that contains the required packages installed to build and test MicroPython, then mount the MicroPython source code directory as a volume and run the same test commands in a docker container using that image.

[Here is an example](https://github.com/pybricks/pybricks-micropython/blob/ee91f62848c383511d03dab9206495251ed71edf/.github/workflows/build.yml#L85-L90) of one of the repos I mentioned. Once the docker container is created, it is basically a matter of taking the existing CI script an adding `docker exec --tty` in front of each command.

stinos, 2021-05-17:
> Though I manually tested the changes using a MIPS

Off-topic, but are you also the starter of this topic: https://forum.micropython.org/viewtopic.php?f=12&t=10443 ? And in any case: if you have some time can you please test [this ](https://forum.micropython.org/viewtopic.php?f=12&t=10443#p57865) piece of code shown to see if it can be reproduced, and verify whether the [suggested workaround](https://forum.micropython.org/viewtopic.php?f=12&t=10443#p57874) works? Thanks!

dlech, 2021-05-17:
FWIW, I also just came across this: https://github.com/micropython/micropython/blob/47e6c52f0c2bf058c5d099dd2993192e0978e172/tools/ci.sh#L181

The qemu-arm port is already doing a big-endian build. But doing a big-endian unix coverage build would probably still exercise quite a bit more code.

dpgeorge, 2021-05-18:
Thanks for the report, these bugs need to be fixed.

Would be great to set up a CI build that uses big endian, probably via QEMU MIPS.

dpgeorge, 2021-06-08:
Fixed by 525a920ca5f284b7eb5875550ae0febcd1714a06

closed 2021-06-08 by dpgeorge