micropython/micropython #7274: ESP32 exception BREAK instr when running with async v3

bhcuong2008, 2021-05-18:
Hi,

I've reported my issue at micropython-async. I think that my issue may be relevant to MicroPython async, so I post it here.
https://github.com/peterhinch/micropython-async/issues/60

I've faced issue with async code. I've tested 2 cases: loop with synchronous code, and loop with async code V3 as follow on ESP32 without PSRAM. I use the latest MicroPython version 1.15.

### 1. Synchronous code:
When running this function, it could run well 24/7 without any exceptions. Sleep in 1ms.

```
def _ctask_fn(arg):
   try:
      while True:
         _ccall(arg['_cfn'], arg)
         time.sleep_ms(arg['sleep'])
   except Exception as e:
      print('[_ctask_fn] %s' % str(e))
   finally:
      return
```

### 2. Async code
Also sleep in 1ms. But run in tens of seconds, or some minutes (2, 3), it occurs exceptions. It's rather random. Most of them were 
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: BREAK instr

arg['_cfn'] is just a callback function, do nothing for test.

If sleep in 50 or 100ms, it's running rather long without exceptions. I've not tested with several hours/days.

```
async def _ctask_afn(arg):
   try:
      while True:
         _ccall(arg['_cfn'], arg)

         if 'sleep' in arg:
            await asyncio.sleep_ms(arg['sleep'])
         elif 'event' in arg:
           await arg['event'].wait()
           arg['event'].clear()
         elif 'tsf' in arg:
            await arg['tsf'].wait()

   except Exception as e:
      print('[_ctask_afn] %s' % str(e))
   finally:
      return
```

I really dont know why. It's just a normal/minimal async code. Pls guide me how to debug it. I think that problem is at async scheduler side.

Thanks.

peterhinch, 2021-05-20:
Can anyone replicate this? I couldn't on the ESP32 reference board, where it ran for > 2 hrs.

dpgeorge, 2021-05-21:
I can not reproduce this using a TinyPICO and latest master (running for about 2 hours).

It's worth noting that the original report is running with a custom integration of MicroPython in an Arduino code base with IDF v3.

dpgeorge, 2021-05-31:
Closing, see https://github.com/peterhinch/micropython-async/issues/60#issuecomment-851104705 for more details.

closed 2021-05-31 by dpgeorge