micropython/micropython #7281: [SSD1306 library issue] Display incorrect on non 128*64 pixel OLED?

water5, 2021-05-18:
I try to use ssd1306 library to 72*40 pixel OLED.
But it looks like can't display after x-pixel 100.

`import ssd1306`

`oled = ssd1306.SSD1306_I2C(72, 40, i2c)`
`oled.fill(1)`
`oled.show()`
[image]

`oled.text('123456789', 0, 0)`
`oled.show()`
[image]

`oled.text('12345', 28, 0)`
`oled.show()`
[image]

mcauser, 2021-05-19:
I don't have one of these displays to test on, but after reading the differences between ssd1306.py and u8g2, I found some config changes to try.

After initialising your display, try running:
```
oled.write_cmd(0xDA)  # SET_COM_PIN_CFG
oled.write_cmd(0x12)

oled.write_cmd(0x81) # SET_CONTRAST
oled.write_cmd(0xAF)

oled.write_cmd(0xD9) # SET_PRECHARGE
oled.write_cmd(0x22)

oled.write_cmd(0xDB) # SET_VCOM_DESEL
oled.write_cmd(0x20)
```

Then try to write some text() again and show().

water5, 2021-05-19:
The issue still exist.

72*40 OLED Specification:
[https://github.com/water5/MicroPython_temporary/blob/master/7240-OLED.pdf](https://github.com/water5/MicroPython_temporary/blob/master/7240-OLED.pdf)

 I follow page 14 to setup, but not work.

water5, 2021-05-20:
I use CircuitPython Adafruit_SSD1306, seems fine,
[https://github.com/adafruit/Adafruit_CircuitPython_SSD1306/blob/master/adafruit_ssd1306.py](https://github.com/adafruit/Adafruit_CircuitPython_SSD1306/blob/master/adafruit_ssd1306.py)

`from board import SCL, SDA`
`import busio`
`import adafruit_ssd1306`

`i2c = busio.I2C(SCL, SDA)`
`oled = adafruit_ssd1306.SSD1306_I2C(72, 40, i2c)`

`oled.fill(1)`
`oled.show()`
[image]

`oled.text('123456789ABC', 0, 0, 1)`
`oled.show()`
[image]

yoyojacky, 2021-06-04:
when I run this game: ` https://raw.githubusercontent.com/printnplay/Pico-MicroPython/main/picoinvaders.py` 
it appears this error, please help.. I am pretty sure the Pin connection is correct.
Pot = ADC(26)
speaker == PWM(Pin(18))
i2c = I2C(0)
>>> %Run -c $EDITOR_CONTENT
Traceback (most recent call last):
  File "<stdin>", line 20, in <module>
  File "/lib/ssd1306.py", line 114, in __init__
  File "/lib/ssd1306.py", line 36, in __init__
  File "/lib/ssd1306.py", line 71, in init_display
  File "/lib/ssd1306.py", line 119, in write_cmd
OSError: [Errno 5] EIO

mcauser, 2021-06-17:
I bought one of these 72x40 displays and can replicate the problem.
Will create a fix PR shortly.

mcauser, 2021-06-17:
@water5 There's 2 things that needed fixing.

Needed to select the internal IREF
oled.write_cmd(0xAD)
oled.write_cmd(0x30)

Needed to update the `show()` to use a column offset of `28` instead of `0`.
64px wide displays had been hard coded to 32, which is exactly 1/2 of the remaining free pixels. (32+64+32=128)
The 72px display needed 28px padding each side to centre the display (28+72+28=128)

dpgeorge, 2021-06-17:
Fixed by bc7822d8e95c40a9d5e403fd22c82b1bbad53b8b

closed 2021-06-17 by dpgeorge