micropython/micropython #7286: RP2 and others: MP version number disappeard.

robert-hh, 2021-05-19:
When building new firmware, the MP version number in the startup message is gone. I get something like:
`MicroPython 7408ca1d7 on 2021-05-19; Raspberry Pi Pico with RP2040`. 

When running 'git describe --dirty --always --match v[1-9].*' on
I get `7408ca1d7`

Is that me having that effect only? Git tells me, that the repository in on par with the web site.

dlech, 2021-05-19:
Did you fetch tags locally?

    git tag --list

robert-hh, 2021-05-19:
Yes, the tags are present. Just the mechanisms which create the Version Label during the build fails now. No change my process, just running make.

robert-hh, 2021-05-19:
It's not limited to the RP2 board. It fails with the other boards too.

dpgeorge, 2021-05-19:
It works fine for me:
```
$ git describe --dirty --always --match 'v[1-9].*'
v1.15-126-g7408ca1d7
$ strings build-PICO/firmware.elf|grep v1.15
MicroPython v1.15-126-g7408ca1d7 on 2021-05-20; Raspberry Pi Pico with RP2040
v1.15-126-g7408ca1d7
v1.15-126-g7408ca1d7 on 2021-05-20 (GNU 10.3.0 MinSizeRel)
```

robert-hh, 2021-05-20:
You're right. On a fresh clone it works again. Hmm ... But how to repair the other one with all my branches?

closed 2021-05-20 by robert-hh