micropython/micropython #7289: Crash micropython in pairheap.h

isgmini, 2021-05-20:
Hello Team,

Can you help us to identify why this crash appears or what we can do to avoid it?

```
assertion "node->child == NULL && node->next == NULL" failed: file "../../py/pairheap.h", line 88, function: mp_pairheap_push
abort() was called at PC 0x40178577 on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x40095138:0x3ffce620 0x4009556d:0x3ffce640 0x40178577:0x3ffce660 0x400ef1c2:0x3ffce690 0x400e4136:0x3ffce6b0 0x400dfe71:0x3ffce6e0 0x400dffad:0x3ffce700 0x400ee07f:0x3ffce720 0x400e41f8:0x3ffce7c0 0x400dfe71:0x3ffce7f0 0x400edff9:0x3ffce810 0x400e41f8:0x3ffce8b0 0x400dfe71:0x3ffce920 0x400dffad:0x3ffce940 0x400ee07f:0x3ffce960 0x400e41f8:0x3ffcea00 0x400dfe71:0x3ffcea70 0x400dfe9a:0x3ffcea90 0x400dff0b:0x3ffceab0 0x400eb251:0x3ffceb40 0x400eb597:0x3ffceb70 0x400e0032:0x3ffcec50 0x400ee339:0x3ffcec90 0x400e41f8:0x3ffced30 0x400dfe71:0x3ffced90 0x400dfe9a:0x3ffcedb0 0x40107d1a:0x3ffcedd0 0x40107ede:0x3ffcee60 0x400f79bd:0x3ffceeb0 0x40095b61:0x3ffceee0
```

looking into the pairheap.h:

```
// Push new node onto existing heap.  Returns the new heap.
static inline mp_pairheap_t *mp_pairheap_push(mp_pairheap_lt_t lt, mp_pairheap_t *heap, mp_pairheap_t *node) {    
                 assert(node-&gt;child == NULL &amp;&amp; node-&gt;next == NULL);    
                 return mp_pairheap_meld(lt, node, heap); // node is first to be stable
}
```

What means "push a new node"? This si related with uasyncio? (We are using uasyncio)

Thanks for help us.

stinos, 2021-05-20:
Can you show the minimal code which reproduces this?

isgmini, 2021-05-20:
Hello stinos,

Unfortunately, we are not able to reproduce it. It appears randomly and not often. After 3-5 hours of running our firmware. The ESP32 reboots.

isgmini, 2021-05-20:
We need information about what is the job of the pairheap.h. Does it to be related with the Garbage Collector? We always enable it in automatic (gc.enable())

dpgeorge, 2021-05-20:
This may be related to #7274.

> This is related with uasyncio?

Yes, each task in uasyncio is a node/element of a pairing heap, which is a data structure used to efficiently find the next task to run.

Can you decode the backtrace?  (you can use idf.py monitor to do this)  Or provide the .elf file that ran and produced those backtrace addresses?

bhcuong2008, 2021-05-21:
as in #7274 , during testing with asyncio, some exceptions related to pairing heap too.

isgmini, 2021-06-09:
Hello Team,

After a lot of efforts, we finally located the 'line' that is causing the crash:

`if a[-1][0] not in self.want_dns_update:`

It seems that,  if  a=None , instead of throw an exeption, running under uasyncio, micropython crash with:

_assertion "node->child == NULL && node->next == NULL" failed: file "/home/micropython/micropython-autobuild/py/pairheap.h", line 88, function: mp_pairheap_push_

Hope that this helps to improve the micropython code.

Right now, we are testing our code, with this line, to avoid the exeption:

` if a and a[-1] and (a[-1][0] not in self.want_dns_update):`

Best Regards, Joan

alessionossa, 2022-01-15:
I am facing the same issue and tried to extract only the code that generates the error from my ESP32 microcontroller codebase but I got a quite big result for "minimal code to reproduce the issue".

Note that the device is connected to WiFi (through WLAN).

**main.py**
```python
import _thread
from test2 import startNatPunch

try:
    # This is an 'secondary' thread that I want to run 'in background' to do NAT hole punching
    natPunchthread = _thread.start_new_thread(startNatPunch, ())
    
    # This contains the device main functionality's code
    exec(open("test.py").read(), globals())
except KeyboardInterrupt:
    print("Stopped with keyboard")
```

**test.py**
```python
import uasyncio
import json
import socket

async def blink(off_ms):
    try:
        while True:
            await uasyncio.sleep_ms(off_ms)
    except uasyncio.CancelledError:
        print("Ended blink")

# Function to log measurements to a remote server
def logToServer():
    serverName = '1.2.3.4' # The server doesn't need to respond to reproduce the issue
    serverPort = 12011

    try:
        clientSocket = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
        clientSocket.settimeout(5)

        clientSocket.sendto("sample record".encode('utf-8'), (serverName, serverPort))
        clientSocket.close()
    except Exception as sendUdpError :
        print(f"Error logging data to server: { sendUdpError }")

async def main():
    while True:
        # This is used to show that there's ongoing activity on the device
        blinkWorkingTask = uasyncio.create_task(blink(250))

        logToServer()

        await uasyncio.sleep(0.2)
        blinkWorkingTask.cancel()
        await uasyncio.sleep(5)

uasyncio.run(main())
```

**test2.py**
```python
import network
import struct
import socket
import uasyncio
import gc

# Helper function, parse IPv4 data
def parseudppacket(data):
    strLen = len(data) - 28
    # https://en.wikipedia.org/wiki/IPv4#Header
    unpacked = struct.unpack(f"!BB3HBBHII4H{strLen}s", data)
    return unpacked

# Helper function, converts int to IPv4 string
def intToIPv4(int):
    ipData = struct.pack("I", int)
    components = struct.unpack("4B", ipData)
    return f"{ components[3] }.{ components[2] }.{ components[1] }.{ components[0] }"

# Helper function, generates UDP packet header data from provided informations
def make_udp_header(srcprt, dstprt, datal):
    return struct.pack(
        "!4H"   #Source port, Destination port, Length, Checksum
        , srcprt, dstprt, datal+8, 0)

# Helper function, generates UDP packet data from provided informations
def makeudppacket(srcPort, dstPort, data):
    uh = make_udp_header(srcPort, dstPort, len(data))
    return uh+data

# Return the first packet that has device's IP Address as destination
async def _getNatPunchResponse(infoClientSocket):
    try:
        foundPacket = False

        while not foundPacket:
            message = infoClientSocket.recv(2048)
            dataExtracted = parseudppacket(message)
            ipDest = intToIPv4(dataExtracted[9])
            currentWlanIp = network.WLAN(network.STA_IF).ifconfig()[0]
            print(f"Received packet for { ipDest }")
            if (ipDest == currentWlanIp):
                foundPacket = True
        
        return dataExtracted
    except uasyncio.CancelledError:  # Task sees CancelledError
        print('Trapped cancelled error.')
        return None
    except Exception as sendUdpError:
        print(f"Error receiving connection info from server: { sendUdpError }")

async def natPunchDaemon():
    serverName = '1.2.3.4'
    serverPort = 12015

    while True:
        infoClientSocket = None
        
        # Send data to server
        try:
            messageStr = "dummy packet"
            messageData = messageStr.encode('utf-8')
            infoClientSocket = socket.socket(socket.AF_INET, socket.SOCK_RAW, socket.IPPROTO_UDP)
            infoClientSocket.settimeout(0)

            serverPacket = makeudppacket(8266, serverPort, messageData)
            infoClientSocket.sendto(serverPacket, (serverName, serverPort))

            print("Connection info sent")
        except Exception as sendUdpError :
            print(f"Error sending connection info to server: { sendUdpError }")

        dataExtracted = None
        try:
            # This is a possible workaround for https://github.com/micropython/micropython/issues/7386
            await uasyncio.sleep(0.2)
            dataExtracted = await uasyncio.wait_for_ms(_getNatPunchResponse(infoClientSocket), 1000)
        except uasyncio.TimeoutError:  # Mandatory error trapping
            print('Response maximum time exceeded (timeout)')

        if dataExtracted is not None:
            print(f"Received string data: { dataExtracted }")

        if infoClientSocket is not None:
            infoClientSocket.close()
        
        await uasyncio.sleep(6)

def startNatPunch():
    uasyncio.run(natPunchDaemon())
```