micropython/micropython #7295: Keep upip working when the package has published several files per release

aivarannamaa, 2021-05-22:
Currently `upip` doesn't consider distribution packages, which have more than one file per release (https://github.com/micropython/micropython/blob/a7a9f2fe897d2c4e3c18618c996460224df08114/tools/upip.py#L201).

In some cases the publishers might find it useful to provide different content for different targets under same distribution name. For example, only small part of https://pypi.org/project/Adafruit-Blinka/ is relevant on MicroPython target.

I suggest to remove the one file restriction from `upip`, so that it is able to work with specially made `tar.gz` even if there are other assets (eg. wheels) published under same release.

I can make the pull request.

*For the context -- I started to work on a [tool called `minipip`](https://github.com/aivarannamaa/minipip/), which is inspired by [micropip.py](https://github.com/peterhinch/micropython-samples/tree/master/micropip) by @peterhinch, but supports also regular pip-installable packages. If `upip` becomes less strict about the number of files, then (in my vision) the community can start publishing libraries with separate assests for PC (`py3` wheels), CircuitPython (eg. `circuitpython6` and `circuitpython7` wheels) and MicroPython (eg. `micropython1` wheels for installing with pip and `tar.gz` for installing with `upip`).*

*This issue only considers `tar.gz` files, but you might want to look at https://github.com/aivarannamaa/minipip/issues/5 for more ideas to ponder.*

peterhinch, 2021-05-22:
The one caveat is to ensure that an enhanced `upip` is still effective on ESP8266 targets. As I understand it `upip` was significantly optimised to achieve this.

aivarannamaa, 2022-01-07:
Related to #7067

aivarannamaa, 2024-07-12:
Closing, as upip is now history.

closed 2024-07-12 by aivarannamaa